Ultimaker/Cura #8248: 4.7 nozzle size grayed out

lnzmoore, 2020-08-26:
**Application version**
4.7

**Platform**
windows 10

**Printer**
ender 3 pro

**Reproduction steps**
1. installed 4.7, it automatically  started with 6mm nozzle selected (which was what I was using on 4.6x)
2. Dropdown for nozzle size has all grayed out selections cannot select any other size

**Screenshot(s)**

[image]

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

nallath, 2020-08-27:
This is an issue with the sidebar plugin. Please contact the developer of that plugin.

fieldOfView, 2020-08-27:
Is it though? @lnzmoore, can you please test if this also happens with the Sidebar GUI plugin disabled?

nallath, 2020-08-27:
I checked this without the sidebar plugin. The options are selectable there.

fieldOfView, 2020-08-27:
But did you reproduce it *with* the sidebar plugin?

fieldOfView, 2020-08-27:
Personally I suspect the underlying issue here is the same as #8253 and #8229

nallath, 2020-08-27:
I haven't tried it with the sidebar plugin, but i have tried it with a PLA+ profile

nallath, 2020-08-27:
I just tried this with the sidebar plugin installed, but I could also not reproduce it (both with generic PLA and a custom material type)

fieldOfView, 2020-08-27:
Ok, so it probably *isn't* caused by the sidebar plugin, because the sidebar plugin mostly just re-_arranges_ existing UI elements (instead of re-_implementing_ them, which could lead to new issues).

nallath, 2020-08-27:
Anyhow, I do need more information to debug this; Logs & configuration files.

lnzmoore, 2020-08-30:
Hi Jamie,
Sorry, I am new to this github thing and don't really understand it. I just
pressed some buttons to report the problem and here we are. I had no idea
real humans would be interacting with me.
I tried making it think i was back on a .4 nozzle. I uninstalled 4.7,
changed 4.6 to a .4 nozzle (because I think the installer looks to the
current version to decide some things) and reinstalled 4.7 new one came up
as a .4 nozzle, but as soon as I drag and dropped an STL (I tried with
several) it gave me an error report which i sent, and it shut down cura.
so  i have no idea how to get you logs and configuration files. so here is
a copy of the part it shows in a box labeled "error traceback"
---

Traceback (most recent call last):

File
"X:\4.7-exe\build\inst\lib\python3.5\site-packages\cura\CuraApplication.py",
line 1060, in event

File
"X:\4.7-exe\build\inst\lib\python3.5\site-packages\UM\Qt\QtApplication.py",
line 464, in event

File "X:\4.7-exe\build\inst\lib\python3.5\site-packages\UM\Event.py", line
218, in call

File "X:\4.7-exe\build\inst\lib\python3.5\site-packages\UM\Signal.py", line
332, in __performEmit

File
"X:\4.7-exe\build\inst\lib\python3.5\site-packages\cura\CuraApplication.py",
line 1820, in _readMeshFinished

IndexError: list index out of range

---

and this is what it says under "logs"

Thread 0x00000e44 (most recent call first):
  File "C:\Program Files\Ultimaker Cura
4.7\plugins\USBPrinting\USBPrinterOutputDeviceManager.py", line 87 in
_updateThread
  File "threading.py", line 862 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00002464 (most recent call first):
  File "C:\Program Files\Ultimaker Cura
4.7\plugins\RemovableDriveOutputDevice\RemovableDrivePlugin.py", line 61 in
_updateThread
  File "threading.py", line 862 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00000f00 (most recent call first):
  File "threading.py", line 297 in wait
  File "zeroconf\__init__.py", line 2117 in wait
  File "zeroconf\__init__.py", line 1434 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x000008bc (most recent call first):
  File "threading.py", line 297 in wait
  File "threading.py", line 549 in wait
  File "C:\Program Files\Ultimaker Cura
4.7\plugins\UM3NetworkPrinting\src\Network\ZeroConfClient.py", line 81 in
_handleOnServiceChangedRequests
  File "threading.py", line 862 in run
  File "threading.py", line 914 in _bootstrap_inner
[... 236 lines cut ...]
  File "X:\4.7-exe\build\inst\bin\cura_app.py", line 234 in <module>
  File "Console.py", line 24 in <module>
  File "<frozen importlib._bootstrap>", line 634 in
_load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 957 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 968 in _find_and_load
  File "__startup__.py", line 12 in <module>
  File "<frozen importlib._bootstrap>", line 634 in
_load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 957 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 968 in _find_and_load

---

So that is what I could find by guessing. I wasn't sure what configuration
files would be, so here is a snip of what the error report says at the top:
[image: image.png]
So I am going to take another wild guess and attach a file called
"Cura.log" because it seemed to have a lot of info in it and my pc actually
opened that one with the text editor thing when I clicked on it.

Hope this helps? Can I do anything else?
Zeb

nallath, 2020-08-31:
That is a duplicate of https://github.com/Ultimaker/Cura/issues/8230

closed 2020-08-31 by nallath