Ultimaker/Cura #8249: Multiple Crash Reports + Missing Profiles/Materials/Printers

Schokobecher, 2020-08-26:
**Application version**
4.7.0 (AppImage)

**Platform**
Linux x64

**Printer**
Ender-5 Pro

**Reproduction steps**
1. Open Cura
2. Find that not all profiles are shown
3. Try to change the active printer
4. Receive 32-64 instances of crash reports

**Screenshot(s)**
[image]

[image]

**Actual results**
Cura crashes on switching the active printer after switching the material on another one.
Cura doesn't show the same printers as 4.6.1 - instead shows an older set (with unchanged name)
Cura doesn't show all Profiles that are still visible in 4.6.1

**Expected results**
Can switch the printer successfully, seeing all my profiles that were there in 4.6.1, see the printer I added in 4.6.1 under the same names as there.

**Log file**
[cura.log](https://github.com/Ultimaker/Cura/files/5132493/cura.log)

**Additional information**
Traceback / error log:

`Traceback (most recent call last):
  File "/build/4.7/build/inst/lib/python3.5/site-packages/cura/Machines/Models/BaseMaterialsModel.py", line 162, in _update
IndexError: list index out of range
`

nallath, 2020-08-27:
I think this is related to https://github.com/Ultimaker/Cura/issues/8230

Could you share your 4.6 configuration folder?

Schokobecher, 2020-08-27:
Sure thing

[8249_4.6.zip](https://github.com/Ultimaker/Cura/files/5135930/8249_4.6.zip)

Let me know if I can assist any further

nallath, 2020-08-27:
Yep, that's the issue I linked to. Closing this as a duplicate.

closed 2020-08-27 by nallath