Ultimaker/Cura #8250: Dual Extruders Selection Causing Delay

3DStyle, 2020-08-26:
**Application version**
4.7.0 and below.

**Platform**
macOS Catalina (10.15.6)
Macbook Pro 13", 2.3GHz Quad-Core, Intel Iris Plus Graphics 655 1536 MB.

**Printer**
TEVO Tarantula with dual extruders. 

**Reproduction steps**
When I try to enable/disable extruder 0 or extruder 1, Cura becomes slow and unresponsive for few moments.
This is causing the user (me) to double, or even triple click the checkmark box since I'm not sure if it was clicked or not, and causing mixed results. 

**Screenshot(s)**
Youtube video to see this issue:
https://youtu.be/P-IDhu8Sfbo

**Actual results**
The checkbox mark gets too much time to be selected.

**Expected results**
Clicking should apply changes instantly. 

**Project file**
This happens with every STL file.

Ghostkeeper, 2020-08-28:
It's true. It's been like this for a long time, basically since the checkbox got introduced. It takes a full second for me at 3.5GHz clock speed.

The problem is that there's a lot to update. It needs to update the disallowed polygons on the build plate (and re-check if any objects on the build plate collide with it). And it needs to re-evaluate all settings that use a resolve function. Those are probably the two most time-consuming parts.

Still I expect that there is room for improvement there. I'd expect it to update all settings blindly now, while only those with resolve functions need to be updated. And there are some inefficiencies with updating anything on the build volume. Maybe it's calculating that twice too because after updating the settings due to the resolve functions it could be updating the build volume a second time. Profiling will tell if we have such bugs still in there.

nallath, 2020-09-04:
I've done a fair bit of profiling on this, but there must be something in the configuration here that is causing this, because I'm having the hardest time fully reproducing it.

If I profile the `setExtruderEnabled` function, it's about 0.15ms (which isn't fast, sure, but it's far from the slowness that we see in the original report. 50% of that time is spent on the `updateNumberExtrudersEnabled` function, which is the one that acutally changes the value, which causes the settings to be re-evaluated (due to them becoming disabled or their actual value changing). 

I'm going to poke around this a bit more since it is an interesting challenge (and hey, the interface _really_ shouldn't be that slow), but I could use a project file (so I can at least rule out if it's a setting issue or not!)

Ghostkeeper, 2020-09-07:
Maybe you're measuring differently. I've been trying with a freshly added Ultimaker 3. But I'm measuring in the GUI from the moment that I click to the moment that I see the checkbox grey out (because you can't disable both extruders). This takes about a second for me. In this period it does 1 screen update about halfway when it switches the active extruder, but it essentially freezes the GUI the whole time.

Here is a project file, but I don't think it'll make much of a difference: [Yoda.3mf.zip](https://github.com/Ultimaker/Cura/files/5183194/Yoda.3mf.zip)

nallath, 2020-09-07:
I'm just looking at the actual function. Since the signals are emitted on the same thread, the time it takes to handle them should be taken into account.

closed 2024-12-07 by github-actions[bot]