Ultimaker/Cura #8252: Prime blob

jaques-strap, 2020-08-27:
**Application version**
4.7

**Platform**
Windows 10

**Printer**
Ender 3

**Reproduction steps**
1. all prime blob settings are marked as "This setting has been hidden by the active machine and will not be visible"
2. I set up different printer profiles and still get the same notation

**Screenshot(s)**
[image]

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

Ghostkeeper, 2020-08-27:
The prime blob executes the `G280` g-code command in the place where it needs to prime. This g-code command is not supported by the Ender 3, so this is expected behaviour.

The Ender 3 prints a line as part of its start g-code in order to prime.

closed 2020-08-27 by Ghostkeeper

jaques-strap, 2020-08-27:
But in 4.6 it worked fine though🤷🏻‍♂️

Ghostkeeper, 2020-08-28:
Really, the Ender 3 printer is doing this movement? https://youtu.be/F470kQ8im80?t=4614

In Cura 4.6 the setting should also be disabled. And from what I can see, it is disabled there too, in the same way:
[image]

jaques-strap, 2020-08-28:
Yeah, I reinstalled 4.6 last night & saw that too.... Which makes it even
more weird since the only thing I've ever changed was simply upgrading to
4.7 & all of a sudden I get no control over the blob. If I could just turn
it OFF I'd be satisfied

Ghostkeeper, 2020-08-28:
It's not actually creating `G280` in the g-code, is it? Do you have a project file for me?

jaques-strap, 2020-08-28:
I have no idea lol. I'm still pretty new to all this. & no, I don't have a
file I can give right now. I'm out of town for a few days. Thank you very
much for all your help though, I'll go some googling when I get back home &
let you know if I still need help