Ultimaker/Cura #8254: missing the view of build plate touching surface parts of the objects

UM2EXTENDED, 2020-08-27:
**Application version**
4.7

**Platform**
macOS

**Printer**
UM2E+

Normally you can see the parts of the objects that will have contact to the surface of the build plate if you look from underneath the build plate. Now you need to slice it again and again and use the layer view to check wich parts will touch the surface for the first layer.

nallath, 2020-08-27:
I don't understand what you mean. There hasn't been a feature that shows you what parts are touching the buildplate, so I'm not quite sure what it is that you saw.

UM2EXTENDED, 2020-08-27:
This is what I mean:

[image]

If you look from underneath you don't know the surface touching parts of objects any more.

fieldOfView, 2020-08-27:
That dark grey "shadow" does not show what parts of objects touch the buildplate. Never has either. It is a projection of the model down to the buildplate. You can see that on the top, where the cylindrical overhanging part definitely does not touch the buildplate, but is included in the shadow. Even the hole in the grey shadow you point out is much smaller than the actual part that does not touch the buildplate. 

It could be interesting (and fairly easy) to add a differently colored indication of faces touching the buildplate.

UM2EXTENDED, 2020-08-27:
No, sorry, that not what I mean and what Cura provides for all version before.

4.6.1
Only the small cylindrical part is touching the build plate. Dark gray doesn't touch the build plate. Easy to see and identify.

4.7.0
There is no way to identify that only the small cylindrical part is touching the build plate. This is impossible to use if you need to lower object to for example -0,7mm to get the above dark grey surface to touch. In this version it's not possible to identify any surface touching part of any object.

fieldOfView, 2020-08-27:
Ah, I misread the perspective on the model.

nallath, 2020-08-27:
Well, the "feature" was never intended as such. It's an unintended consequence of the way we used to render the shadows. 

In one of the optimizations, I removed the backface cull, which is why it's not being rendered from below anymore.

UM2EXTENDED, 2020-08-28:
That's a problem because then Cura is useless for our business usage. Now I need to go back to 4.6.1, which is no big deal as this version was very stable, a bit slow to be honest compared to simplify 3D, but stable on macOS.

If this important render view will be added again I can update to a new version.

fieldOfView, 2020-08-28:
If you are willing to replace a file in your Cura installation for now, you could try to download this file:
https://gist.githubusercontent.com/fieldOfView/0f0812545885c5103816a189d9e203b2/raw/17e233123b3b3eeab6fb3020d2ee0df3b9937830/overhang.shader

It replaces the file in the `resources/shaders` inside the AppImage. This patched shader colors parts of models that touch or extend below the buildplate a dark solid purple.

Ghostkeeper, 2020-08-28:
This may have started with some performance optimizations we've made with the rendering of the collision areas. Backface culling is enabled for those collision areas though (always has been, but it seems that it's only now visible). And maybe it's a case of an [unintended workflow](https://xkcd.com/1172/).

closed 2020-09-01 by rburema

nallath, 2020-09-01:
@UM2EXTENDED In case you didn't realise it; we added something like this based on the suggestion of @fieldOfView. See the image of the new color in action!

[image]

fieldOfView, 2020-09-01:
In case it is not clear from the image: the color of elements touching the buildplate or extending below it gets inversed (ie: yellow becomes blue)

commit 8dbbe2bd (2020-09-01): Use different colors below buildplate.

fixes #8254