Ultimaker/Cura #8255: 2-in-1 shared heater

cbldwn, 2020-08-27:
Hello,
   I'm running a y-splitter in my bowden tube to my hotend and had been using the shared heater option in the printer settings, which in the new 4.7 looks to be removed. This needs to be added back as every time the hotend does a tool change its dropping  the temperature to 195 even tho I have the standby temp set at 212. 

Thanks,
     Craig Baldwin

fieldOfView, 2020-08-27:
As far as I can see, the option is still there. Perhaps I could reproduce your issue if you had filled in the issue template.

cbldwn, 2020-08-27:
It is not still there, there isn’t even a box to check for it.
[image]

fieldOfView, 2020-08-27:
I am sorry, you are right. The machine I checked with was still running Cura 4.7 beta. It does seem the feature has been temporarily removed.

The "Shared Heater" feature is temporarily disabled because its implementation is incomplete. Printers with multiple filaments going into one nozzle will keep the inactive filaments retracted at the start of a print. However CuraEngine assumes that all filaments start at the nozzle tip. So it'll start printing the second filament without unretracting it.

See: https://github.com/Ultimaker/Cura/issues/8148

cbldwn, 2020-08-27:
Yes that’s not a problem, don’t use cura’s tool change. Put the filament load command in the tool start and the retract in the end code and then everytime Cura calls a tool it will load the filament before before continuing to print and unload the filament before loading the next extruder. I’ve have been using this setup without any problems until the update.

fieldOfView, 2020-08-27:
If you don't mind hacking Cura a bit, you could replace `C:\Program Files\Ultimaker Cura 4.7\plugins\MachineSettingsAction\MachineSettingsPrinterTab.qml` with this version: https://raw.githubusercontent.com/Ultimaker/Cura/c7e6553dbfb5cdcfbe0ba0e7b263d64062fe0952/plugins/MachineSettingsAction/MachineSettingsPrinterTab.qml

cbldwn, 2020-08-27:
I don’t mind, I’ve already written my own .json files for my printer to make upgrading quicker and easier for me. What is more of an issue with the prime tower is the order in which the prime tower prints when I was setting up my 2-in-1-out splitter and trying to get it working I noticed a lot of times after a tool change instead of printing the purge tower it would start printing the part. And then go to the purge tower. Because of this I went away from the purge tower and started using a purge bucket, not only did it eliminate my issue of printing the part before purging but it also dropped my print time from 45 min to 25. The black and gold one was printed with the prime tower and the gold and copper with my purge bucket for comparison.

[image]
[image]

fieldOfView, 2020-08-27:
That sounds like another issue. Issues here should be one specific problem. It seems your original problem is solved for now, so you can close this issue. 

I just felt bad about saying the checkbox was still there while it wasn't, so I wanted to give you a workaround. I know very little about shared heaters and prime towers otherwise.

cbldwn, 2020-08-27:
The issues I mentioned with the prime tower would probably part of the reason why this was disabled in the first place. Another work around to the original filament un-retract problem is suppose to be if running marlin, to enable firmware retract and have Cura output the g10 & g11 for normal retraction and let the firmware handle the retraction for the tool changes. Thanks for the information on where I would have to make the change. I was able to skim through the code and found the comment I had to move up to re-enable the shared heater check box. Photo’s of the change are attached so if someone else has the problem and wants to make the change back as well. 
[image]
[image]

closed 2020-08-27 by cbldwn