Ultimaker/Cura #8257: Conditional OuterBeforeInnerWalls Option

daT4v1s, 2020-08-27:
**Is your feature request related to a problem? Please describe.**
OuterBeforeInnerWalls is less suitable for overhangs but if most of the part does not contain overhangs,
there may be slightly better quality/accuracy esp for measuring devices

it may not be optimal if most of the part does not contain overhangs, but part of it does
[image]
notice the bottom round (there may be other better examples)

[image]
(SettingsGuide plugin)

[image]

**Describe the solution you'd like**
For wall segments that are overhangs, print with default Inner before outer,
perhaps, same for adjacent layer (less inconsistency artifacts)

it should not affect other parts that are vertical or sloped inwards
but it does not have to be a hard-cut,
perhaps extend default-ordering any line segment that for some length,
maybe up to corner/seam or as a whole if there are side effects

but question of  how feasible this is to implement,
and how to partition the problem into layers/parts

**Describe alternatives you've considered**
(A clear and concise description of any alternative solutions or features you've considered. Again, if possible, think about why these alternatives are not working out.)
one alternative is using a cutting modifier mesh, but occasionally there are minor artifacts at the boundary e.g. extra-ironing

besides, extra prep is required, instead of being something 
integrated and automatic

**Affected users and/or printers**
(Who do you think will benefit from this? Is everyone going to benefit from these changes? Or specific kinds of users?)
basically anyone who prints parts that have to be accurate and only some of the model contains overhangs

**Additional context**
(Add any other context or screenshots about the feature request here.)
minor question: note that outer-walls and inner walls aren't printed at the same speed, any implications, 
can the overhang angle for inner/outer first be tuned?, and to what amount..

viper0078, 2020-08-29:
Hi.
What kind of movement do you expect that option to do if there are areas with and without overhangs in the same layer?
The left and right sides of the diagram below correspond to this.
[image]
It is not possible to change the on/off status of the options for the right and left sides of the printout. If you think the left side shape is more important, you can turn it off, and if you think the right side is more important, you can turn it on.
The slicer cannot determine that much.

daT4v1s, 2020-08-30:
for those cases if a Wall (line) is contiguous to to a overhang
perhaps, in those cases print 
the inner wall before out walls.

Ghostkeeper, 2020-09-01:
I think this would cause some unexpected results for the user. Most notably, banding. Take this example of a print:
[image]
Halfway across the layers there is this tiny button, which causes overhang. That makes the entire layer swap from printing the outer wall first, to printing the inner wall first. With the Ultimaker printers' default profiles this makes a difference of about 0.1mm in the thickness of the print. It can be more than that too, with an aggressive Outer Wall Overlap setting or a line width that's far below the nozzle size. So you'll get an edge around the entire border of the print where the wall is protruding 0.1mm farther out, every time it switches.

So I'd predict it does more harm than good if you make the order switch halfway across the print. If the horizontal dimensions are inaccurate throughout the entire print, you can at least compensate for it. If it switches it'll be inaccurate either way.

daT4v1s, 2020-10-06:
it seemed kind of obvious this might be a problem, 
what about limiting  the frequency amount of switches  that occur

another separate horizontal-expansion setting to compensate for those switches 
may be needed to suppress those inconsistencies

using outer before inner more
might make lines  bond together more tightly 
(something for inner lines to press against, vs outer lines with open-space on one side)