Ultimaker/Cura #8258: Conical support did't work on delta printers

MrStump, 2020-08-28:
**Application version**
4.6 and 4.7

**Platform**
windows 10 and MAC

**Printer**
delta printer, any. "Buildplate Elliptic", "Origin at center"

**Reproduction steps**
Create new custom delta printer, enable "Conical Support"

**Screenshot(s)**
below

**Actual results**
No Conical Support

**Additional information**
I have few printers, on CoreXY profile, Ender profile "Conical supports" work well.
On delta printer don't
ALSO when i rise model on any distance above the buildplate and slice it on NON-Delta printer - Cura build support to the model as it should.
when i rise model above the buildplate on Delta printer and slice - Cura show me ghost model on it's position where i rise it, but slice from the table. This happens ONLY when "Conical Supports" enabled. 
This was checked on few different PC's
[image]

fieldOfView, 2020-08-28:
See the setting "Remove Empty First Layers". Apparently it is set for your Delta printer(s), and not for your CoreXY printer(s)

MrStump, 2020-08-28:
> See the setting "Remove Empty First Layers". Apparently it is set for your Delta printer(s), and not for your CoreXY printer(s)

sorry, my bad. it was **enabled** by default, and it became invisible when support is **enabled**, but it still did't generate **Conical Support**

Ghostkeeper, 2020-09-01:
It seems to work fine for me. Here is a cylinder with a rod in the middle to keep it upright:
[image]

Or just a floating cylinder (with "drop models to build plate" disabled):
[image]

Do you have a project file by which we can reproduce the problem? It's requested in the issue template.

MrStump, 2020-09-01:
here is test STL, disabled and enabled conical support

enbled support, disabled **Conical**
[image]

enabled support, enabled **Conical**
[image]

and both 3mf files
[conical_test.zip](https://github.com/Ultimaker/Cura/files/5159222/conical_test.zip)

Thanks.

Ghostkeeper, 2020-09-03:
Indeed, it seems to depend on the Origin at Center setting, not necessarily on the build plate shape.

Ghostkeeper, 2020-09-07:
Fixed for 4.8.

closed 2020-09-07 by Ghostkeeper