Ultimaker/Cura #8259: Why tool0 prime more than tool 1?

tphan26363, 2020-08-28:
I am adjusting the amount of prime and try to save some filament and time in tool changer,
  How ever when I reduce the prime towel size to 5 mm.  It seem that the tool1 is not even prime any filament while tool 0 is prime OK.  Apparently by some reason the prime amount for tool1 is significantly less than tool 0.  is there any way to make them equal?
Thanks in advance

Ghostkeeper, 2020-09-01:
I'm unsure what settings you are changing. There are several different ways to prime. Maybe you are adjusting the extruder start g-codes? Maybe you're adjusting the Prime Tower Minimum Volume? Maybe you're adjusting the skin lines or brim width? In any case, make sure that you adjust them on all of the tabs (or right-click on the setting and select "copy value to all extruders").

Could you fill in the [bug template](https://github.com/Ultimaker/Cura/blob/master/.github/ISSUE_TEMPLATE/bug-report.md) for us, which you graciously deleted when you created that ticket? It helps a lot to see what's going wrong. In particular the project file and the version number are important.

tphan26363, 2020-09-01:
The parameter is under Dual extrusion setting:
-Enable Prime Tower
-Prime Tower size
I adjust these for both extruder at exact same values.  But it seem that the prime volume is different?

Ghostkeeper, 2020-09-03:
There is also Prime Tower Minimum Volume. Please ensure that this setting is also equal.

The minimum volume will also be rounded up to complete the circle. Since the circles are nested in each other (with different radii), the lines will never have exactly the same volume. But the Prime Tower Minimum Volume should be held at least.
Is that what you meant? The total volume of the prime tower lines is not the same?

tphan26363, 2020-09-03:
Yes Prime Tower Minimum Volume =4 but I think the 2nd head almost  totally skip it( 1 dot and that's it).  I think I can resolved by setting Prime tower size >=8mm so more of the filament will be primed on T1.  How ever the different of prime volume between two hot end is too much.  The T0 was like 99 % and the T1 just like 1%.  I think it should be better than that even if they are not equal in amount they should be primed enough per head

Ghostkeeper, 2020-09-04:
Note that Prime Tower Minimum Volume is also a dual-extruder setting.

I'll ask again, do you have a project file for us? Fill in the template, please? It would be so much easier to tell what's going wrong then.

tphan26363, 2020-09-05:
Project File attached
[CFFFP_loubie_aria_resculpt_base_and_eyes_v1.1.zip](https://github.com/Ultimaker/Cura/files/5177418/CFFFP_loubie_aria_resculpt_base_and_eyes_v1.1.zip)

tphan26363, 2020-09-05:
Attacked is working and not working
[2files.zip](https://github.com/Ultimaker/Cura/files/5177432/2files.zip)

Ghostkeeper, 2020-09-07:
Thanks. The "working" file is a project file, but the "not working" file is plain 3MF. Please use File -> Save Project... (Ctrl+S) rather than "Export" or any other way of saving things.

At least in the "working" file it seems to be working correctly indeed. And also when using the "working" settings on the "notworking" 3MF model.
[image]

closed 2020-09-21 by no-response[bot]