Ultimaker/Cura #8260: (Cura 4.7) Profiles not imported when chnaging nozzle size

Chuggers9, 2020-08-28:
Application version**
4.7

**Platform**
Windows 10

**Printer**
Ender 3

**Reproduction steps**
1. Import printer profiles on the default nozzle (0.4mm)
2. Change the nozzle size to any other setting (i.e 1.0mm)
3. No profiles are stored in CURA
4. Import your own printer profile, a message appears saying import Succesful
5. No profile loaded 

**Screenshot(s)**
Nozzle setting 0.4
[image]

Nozzle setting 1.0mm

[image]

Import profile on 1.0mm nozzle

[image]

After Import

[image]

**Actual results**
No profiles inserted after import.

**Expected results**
Compared to 4.6.2 the profiles never needed to be imported as it was automatically set for each nozzle size.

**Additional information**
Rolled back to 4.6.2 as this was working as intended

nallath, 2020-08-28:
That was a bug in 4.6.2. The profile you're trying to import is based on the "super" quality, which is one with a 0.12 layer height. The 1.0 nozzle doesn't have a 0.12 base profile, only a "draft" profile (0.32 mm layer height).

User profiles are so-called "quality_changes". You can find a bit more information on what that means on our wiki: https://github.com/Ultimaker/Cura/wiki/Profiles-&-Settings#quality-changes

closed 2020-08-28 by nallath

Ghostkeeper, 2020-09-01:
Should it maybe say something in that message about that the profile is not currently visible because one of the extruders doesn't support it?