Ultimaker/Cura #8262: Supports end one layer before the required layer, thus causing the layers to have a poor base.

MikatDude, 2020-08-28:
Supports end one layer before the required layer, thus causing the layers to have a poor base.

I have tried this in GRID, ZIGZAG, CONCENRTIC and a few others, all to the same effect.

When printing relatively small parts especially gears, leaves the gear with poor build and gears are no concentric.

Asterchades, 2020-08-28:
Are you saying they're ending before the gap you've specified, or are you saying they're ending a layer before the actual model starts? Because if it's the latter, it's supposed to do that - the plastic doesn't know nor care if it's being used for a support or a model, so if the two come into direct contact with each other than they'll fuse together. You can of course do this deliberately by changing the Support Z Distance to 0 (default 0.1) but you're going to have to cut the supports off if you do that.

Of course, if you're telling the supports to leave a certain gap (eg 0.1mm, which will be rounded up to the next layer) and it's leaving a bigger gap than it could be a problem. But I have my suspicions you're expecting undesirable behaviour.

Ghostkeeper, 2020-09-01:
Indeed, this is the result of the Support Z Distance and expected behaviour.

If you disagree that it's expected behaviour, please fill in the bug template. It allows us to see what you really mean. In particular the project file is very useful for debugging slicing issues.

closed 2020-09-01 by Ghostkeeper