Ultimaker/Cura #8263: Y and Z scale handles are swapped for support blockers (4.7)

NameOfTheDragon, 2020-08-28:
**Application version**
(The version of the application this issue occurs with.)
4.7

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)
Windows 10

**Printer**
(Which printer was selected in Cura?)
N/A but Creality Ender 3

**Reproduction steps**
1. Select an object
2. Select the "support blocker" tool and add a support blocker.
3. Select the support blocker object
4. Select the Scale/Resize tool
3. Uncheck "uniform scaling"
4. Move the green Y handle - Z scale changes. Move the blue Z handle - Y scale changes.

**Screenshot(s)**
A screen shot adds no value in this instance. 

**Actual results**
Z scale changes when the Y handle is dragged.
Y scale changes when the Z handle is dragged.

**Expected results**
Obviously, dragging the Y handle should change the Y scale and the Z handle should change the Z scale.

**Project file**
Any object (just add a support blocker, for example).

**Log file**
N/A - this is an interactivity issue.

**Additional information**
I don't think there's anything further to add. Y and Z scale handles are just swapped.

fieldOfView, 2020-08-28:
They are only swapped if the model you added them to is rotated.

Duplicate of #5916, which is a duplicate of #4136

fieldOfView, 2020-08-28:
Quoting myself from that issue:

> In the latest release of Mesh Tools, there is a function to "Apply transformation to mesh". This applies the current scale and rotation to the selected mesh(es) and resets their local coordinate systems. After that, non-uniform scaling should apply to the proper axis (until you rotate the model again). This could be a workaround until the issue is properly fixed in Cura.

Ghostkeeper, 2020-09-01:
Closing as duplicate of https://github.com/Ultimaker/Cura/issues/4136, https://github.com/Ultimaker/Cura/issues/3625, https://github.com/Ultimaker/Cura/issues/4309, https://github.com/Ultimaker/Cura/issues/4386, https://github.com/Ultimaker/Cura/issues/5221, https://github.com/Ultimaker/Cura/issues/5189, https://github.com/Ultimaker/Cura/issues/5310, https://github.com/Ultimaker/Cura/issues/5462, https://github.com/Ultimaker/Cura/issues/5599, https://github.com/Ultimaker/Cura/issues/5602, https://github.com/Ultimaker/Cura/issues/5603, https://github.com/Ultimaker/Cura/issues/5872, https://github.com/Ultimaker/Cura/issues/5916, https://github.com/Ultimaker/Cura/issues/5965, https://github.com/Ultimaker/Cura/issues/6148, https://github.com/Ultimaker/Cura/issues/6150, https://github.com/Ultimaker/Cura/issues/6238, https://github.com/Ultimaker/Cura/issues/6370, https://github.com/Ultimaker/Cura/issues/6619, https://github.com/Ultimaker/Cura/issues/6770, https://github.com/Ultimaker/Cura/issues/6829, https://github.com/Ultimaker/Cura/issues/6890, https://github.com/Ultimaker/Cura/issues/6971, https://github.com/Ultimaker/Cura/issues/7102, https://github.com/Ultimaker/Cura/issues/7142, https://github.com/Ultimaker/Cura/issues/7237, https://github.com/Ultimaker/Cura/issues/7388, https://github.com/Ultimaker/Cura/issues/7446, https://github.com/Ultimaker/Cura/issues/7448, https://github.com/Ultimaker/Cura/issues/7582, https://github.com/Ultimaker/Cura/issues/7638, https://github.com/Ultimaker/Cura/issues/7703, https://github.com/Ultimaker/Cura/issues/7768, https://github.com/Ultimaker/Cura/issues/7778, https://github.com/Ultimaker/Cura/issues/7799, https://github.com/Ultimaker/Cura/issues/7946, https://github.com/Ultimaker/Cura/issues/8100 and https://github.com/Ultimaker/Cura/issues/8122.

closed 2020-09-01 by Ghostkeeper