Ultimaker/Cura #8265: [4.7] Automatic model placements sets models out of printable area

Liger0, 2020-08-28:
**Application version**
4.7

**Platform**
Windows 10 x64

**Printer**
Cr10s

**Reproduction steps**
1. Place multiple models on the build plate
2. Automatically place the models

**Actual results**
Some model is places on the bed margin, even though the margins are not a valid printable area and there is space to place them all

**Expected results**
The unprintable area should be considered.

Ghostkeeper, 2020-09-01:
Yeah, we knew about this and improving the arrange functionality is a high-priority task for the next release. It's supposed to be a big feature.

daton89, 2020-10-12:
In my case I put a small model, so it gets placed in the center, then I do multiply and I set `One at a time` as _Print Sequence_ and then I arrange all models. 

The result is that the first model is still in the center and all the other are pleced in line outside of the bed. 

If I arrange before change the _Print Sequence_ options the models are placed too close so then it's not able to print with `One at a Time`

Ghostkeeper, 2020-10-13:
The case for one-at-a-time mode is also reported here: https://github.com/Ultimaker/Cura/issues/8329
It's a bit of a separate issue. The collision areas for objects are wrong in one-at-a-time mode, as can be seen if the "ensure models are kept apart" preference is enabled.

Ghostkeeper, 2020-12-07:
This should be improved significantly with Cura 4.8.

closed 2020-12-07 by Ghostkeeper