Ultimaker/Cura #8268: Grouped models not sliceable if one of group parts is below Z=0

KenLucke, 2020-08-28:
**Application version**
4.7.0

**Platform**
Mac OSX 10.15.6

**Printer**
CR-10S as substitute for CR-10v2

**Reproduction steps**
Load two or more models
Position one model below other modes
Group Models
Lower so that the shortest or lowest model is below Z=0
Attempt to slice

**Screenshot(s)**
Models loaded:
[image]

Models grouped:
[image]

Models sunk below bed:
[image]

Models ungrouped in place to show exposed, ungrouped model will still slice
[image]

**Actual results**
Entire model becomes unsliceable

**Expected results**
A grouped model should be considered for slicing on the basis of the entire group, not just one part.    Sometimes complex models/projects need to be sunk and resliced to continue a failed print or other reasons, and should not have to be disassembled to do so.

BagelOrb, 2020-09-05:
I have the same problem even if the model is not below the build plate. Still I have a feeling it is the same underlying issue.

[image]

[UMS5_vase_bottom.3mf.renamed.zip](https://github.com/Ultimaker/Cura/files/5178327/UMS5_vase_bottom.3mf.renamed.zip)

I seem to be unable to slice anything that's grouped. Can reproduce in Cura 4.7.0 and in Cura 4.6.2

I don't think being below the build plate should ever be enough ground to mark the model as unslicable.

BagelOrb, 2020-09-05:
[gietr_2.zip](https://github.com/Ultimaker/Cura/files/5178329/gietr_2.zip)

Had the same problem if I duplicate and group this model.

Reproducibility is weird. Now all of a sudden I am able to slice. Reproduction conditions are not 100% clear.

nallath, 2020-09-07:
@BagelOrb Could you check the PR?

BagelOrb, 2020-09-07:
Couldn't reproduce the gietr problem, but I did reproduce the vase problem in both 4.7.0 and in your branch.

BagelOrb, 2020-09-07:
So your PR doesn't fix the vase issue.

nallath, 2020-09-07:
Well, that means your feeling is wrong ;) It's not the same underlying issue.

BagelOrb, 2020-09-07:
Shall I make a new issue then?

nallath, 2020-09-07:
Please do!

BagelOrb, 2020-09-07:
Done: https://github.com/Ultimaker/Cura/issues/8336

PR #8302 (2020-12-07): Don't mark children nodes in a group below the buildplate
Fixes #8268
commit 09066f27 (2020-12-07): Don't mark children nodes in a group below the buildplate

Fixes #8268
closed 2020-12-07 by Ghostkeeper