Ultimaker/Cura #8269: Layout assistance

KenLucke, 2020-08-28:
"Arrange Models" is very simplistic, and does not work well a lot of the time.  LuBan, while not a slicer, has a 2d and 3d "nesting" feature that does a bunch of calculations to get the best layout no matter how many and how complex the shapes (https://www.dropbox.com/sh/fxmo3mklf8oa7ex/AABsRVsgkxleHh8lpH3ThVBWa/Doc/English?dl=0&preview=LuBan2D+Draw_Nest.pdf&subfolder_nav_tracking=1)

Could something such as this more exacting arrangement please be added?

[image]

[image]

nallath, 2020-08-31:
An improved version of the nesting algorithm is scheduled for 4.8

closed 2020-08-31 by nallath

Ghostkeeper, 2020-09-01:
Normally we'd keep it open then until it's fixed. However this is a duplicate of https://github.com/Ultimaker/Cura/issues/2781 anyway.