Ultimaker/Cura #8270: Asymmetric (Vertical) Max Speed Accel Jerk Settings

daT4v1s, 2020-08-29:
**Is your feature request related to a problem? Please describe.**
unlike the X and Y axis,
the Z axis has to deal with gravity
have not experimented with this

also, for most printers, the nozzle is at a from top, facing down orientation

**Describe the solution you'd like/propose**
(A clear and concise description of what you want to happen. If possible, describe why you think this is a good solution.)
a way to specify max up&down speed
and if there are is accel/jerk control, likewise.
prepend appropriate gcode to individual moves to tune 

more sophisticated methods may require firmware mods, further outside cura's reach,
also question on how to manage diagonal moves

**Describe alternatives you've considered**
adding custom acceleration/jerk g-code between all those moves is complex
same for modifying speed

**Affected users and/or printers**
(Who do you think will benefit from this? Is everyone going to benefit from these changes? Or specific kinds of users?)
it may affect people who use vertical motion a lot in prints
- Z-hop
- wire-printing (experimental)
- non-planar printing (future?)

**Additional context**
(Add any other context or screenshots about the feature request here.)

Ghostkeeper, 2020-09-01:
So do I understand that you've set your Z hop speed so high that the printer is not able to keep up with it, because of gravity? Like, is the stepper motor losing steps? I would think that acceleration is much more of a limit at that point.

Marlin's default Z acceleration is 100mm/s². With Cura's default Z hop height of 1mm, it would take 0.1s to accelerate to the default Z hop speed of 10mm/s and simultaneously reach the halfway point of 0.5mm Z hop. It then takes another 0.1s to decelerate to 0mm/s again. Assuming the printer can take the acceleration when going up as specced, it may be able to accelerate slightly faster when going down, due to gravity. But we're talking about really tiny differences here, which are more influenced by stepper motor electromagnets than gravity.

To implement this we'd need to break the Z hop move in two and instruct the printer to change its Z acceleration in between the two moves, because the firmware itself doesn't implement asymmetric acceleration. The printer then needs to recalculate its motion buffer (normally 16 moves), and this may cause buffer underruns to appear.

But in the end you may shave off maybe another 0.03s from the Z hop duration... I just don't think it's worth the investment and the risk here, honestly.

I don't expect Wire Printing or non-planar printing to reach the vertical speeds that the Z hop currently does. Wire printing needs to be kind of slow to get the extrusion and cooling right. Non-planar printing will be limited in the slopes it can reach, so it'll likely be more limited by X/Y speeds than Z speeds.

daT4v1s, 2020-09-03:
no, this is just hypothetical, and since I noticeds that vertical max/accel/jerk settings are very low compared to the horizontal ones, and if one of them could be increased and what effect of that would be 
(e.g. 
- horizontal jerk of 20 and  vertical of 0.4
- max horizontal accel of 9000 and  vertical of 100

Ghostkeeper, 2020-09-04:
I think it's low because most printers have a Z screw there which has a very high gear ratio. So while the rotation of the screw may be accelerating just as fast as other motors, the actual Z movement will still accelerate slowly.

GregValiant, 2026-09-10:
I'm going to close this.

closed 2026-09-10 by GregValiant