Ultimaker/Cura #8273: Enlight surface when aligning to build plate

Liger0, 2020-08-29:
**Is your feature request related to a problem? Please describe.**
On the model rotation tool there is the possibility to select a surface to align to the build plate. 
The problem is that CUra doesn't provide neither a wireframe view, neither it enlights the surface you are pointing, so you don't know which surface you are selecting, especially if the model has an high amount of polygons

**Describe the solution you'd like**
Enlight the surface when hovering over it

**Describe alternatives you've considered**
An unplausible wireframe mode to help further

**Affected users and/or printers**
Any

Ghostkeeper, 2020-09-01:
We had this implemented when the Align to Build Plate feature was first implemented. And it looked cool. I don't remember exactly why it got removed, but I think it had something to do with OpenGL 2.0 not supporting it. Do you remember, @rburema?

rburema, 2020-09-01:
The legacy version (OpenGL 2.0 and the like) doesn't have the align-face-to-buildplate at all, so that's what you're thinking of.

I think it had more to do with how the face selection and the object selection interacted, as well as the fact that changing the state of the render for _hovering_ over the object is different from _clicking_ on the object.