Ultimaker/Cura #8275: GCODE not saved correctly

ALSFelix, 2020-08-29:
**Application version**
Cura 4.7

**Platform**
Linux - Ubuntu/Xubuntu 18.04.5

**Printer**
Geeetech A10T 3 into 1

**Reproduction steps**
Copied the GCODE from Cura 4.6.1 for printer and all extruders start AND END gcode.
Closed and reopened the window to verify the code. This was ok.
Closed Cura and when reopened the code for extruders was not all there. only the first four lines.
Repeated this several times and re-downloaded Cura several times to make sure.
It keeps happening.

Thanks

ALSFelix, 2020-08-29:
Forgot to add my GCODE:

;EXTRUDER START G-CODE
M117 Purge E1     ; Message
M211 S0           ; Turn off soft stops
G1 X-17 F6000     ; Go to X=-17 at 100 mm/sec
G92 E0            ; Reset extrusion distance on new tool
G1 E150 F90       ; Extrude 150 mm of filament at 1.5 mm/s on new tool
G92 E0            ; Reset extrusion distance on new tool
G1 E-1 F90        ; Retract 1mm
;CLEAN NOZZLE SEQUENCE
G0 Z3 F1000       ; Raize nozzle to 3mm
G0 X10 Y-6 F6000  ; goto brush
G0 X25 Y-10 F6000 ; Move 15mm to the right and 4mm to the front
G0 X25 Y-6        ; Move 4mm backwards
G0 X10 Y-10 F6000 ; Move 15mm to the left and 4mm to the front
G0 X10 Y-6        ; Move 4mm backwards
G0 X25 Y-10 F6000 ; Move 15mm to the right and 4mm to the front
G0 X25 Y-6        ; Move 4mm backwards
G0 X10 Y-10 F6000 ; Move 15mm to the left and 4mm to the front
G0 X10 Y-6        ; Move 4mm backwards
G0 X25 Y-10 F6000 ; Move 15mm to the right and 4mm to the front
G0 X25 Y-6        ; Move 4mm backwards
G0 X10 Y-10 F6000 ; Move 15mm to the left and 4mm to the front
G0 X10 Y-6        ; Move 4mm backwards
G0 X25 Y-10 F6000 ; Move 15mm to the right and 4mm to the front
G0 X25 Y-6        ; Move 4mm backwards
G0 X10 Y-10 F6000 ; Move 15mm to the left and 4mm to the front
G0 X10 Y-6        ; Move 4mm backwards
G0 X25 Y-10 F6000 ; Move 15mm to the right and 4mm to the front
G0 X25 Y-6        ; Move 4mm backwards
G0 X10 Y-10 F6000 ; Move 15mm to the left and 4mm to the front
G0 X10 Y-6        ; Move 4mm backwards
M211 S1           ; Turn on soft stops
M117 Printing     ; Message
G92 E0            ; Set current extruder position to 0mm
G0 F10000         ; Increase feed rate before next travel

Cura is not saving after 4th line...

MimirDroid, 2020-08-30:
I have this same problem on Windows 10 version.  
Add new printer
paste in custom starting g-code which worked just fine in 4.6
Cura works fine until you close it down.
When you re-open Cura you don't get any error dialog, but the material tab is gone and extruder 1 is not listed under the settings menu.  when you go back into the machine settings of the printer, there is no extruder tab and only the first 4 lines of the starting gcode is in the custom starting gcode box.
Added zip of cura.log and stderr.log
[Cura_logs.zip](https://github.com/Ultimaker/Cura/files/5146155/Cura_logs.zip)

mrzottel, 2020-08-30:
There is actually an equal sign in this line:

G1 X-17 F6000 ; Go to X=-17 at 100 mm/sec

which will cause this. As this is only a part of a non functional argument, just remove it and it should work.

ALSFelix, 2020-08-30:
How did I missed that...
Thanks 
I will try it today and report back...

ALSFelix, 2020-08-30:
I missed the = sign...
Without it it works fine...

Will try to close or delete this...

Thank you

mrzottel, 2020-08-30:
From the logs of MimirDroid I would say it is the same issue.

@MimirDroid : Please check your startup or end gcode for equal signs and remove them. If you are using octolapse and the octolapse script, which contains a lot of equal signs, please either do not use this script until the next version of cura or have a look here: https://github.com/Ultimaker/Cura/issues/8235 for a workaround.

ALSFelix, 2020-08-30:
Ok. When MimirDroid replays i will close this...

MimirDroid, 2020-08-31:
I confirmed this morning that it is the same problem.  I do use octolapse, and after reading that it was likely the equals that was causing the problem, I tried without the octolapse script and Cura started up normally.

nallath, 2020-08-31:
A patch for this is already undergoing testing!

closed 2020-08-31 by nallath