Ultimaker/Cura #8277: Odd issue with curved thin part

Anon0ne, 2020-08-29:
**Application version**
4.7

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)
Win 10 x64

**Printer**
(Which printer was selected in Cura?)
Ender 5

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)
I have tried to turn off optimize walls, increase walls and top/bottoms and many other things that would normally change this. But there are parts of the model which are printing the interior wall broken into columns, and alot of jerky infill style moves causing many artifacts on the exterior skin of model.

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 
https://ibb.co/vcpRkzk
https://ibb.co/PmjkyMH
https://ibb.co/L8YgbZj
https://ibb.co/DPXsX5d

**Actual results**
(What happens after the above steps have been followed.)
https://ibb.co/vcpRkzk
https://ibb.co/PmjkyMH
https://ibb.co/L8YgbZj
https://ibb.co/DPXsX5d

**Expected results**
(What should happen after the above steps have been followed.)
https://ibb.co/qYkvWxN

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

[CE5_fire-tv-remote-cover.zip](https://github.com/Ultimaker/Cura/files/5146059/CE5_fire-tv-remote-cover.zip)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

smartavionics, 2020-08-30:
I tweaked a few settings, is this better?

[CE5_fire-tv-remote-cover-mb.zip](https://github.com/Ultimaker/Cura/files/5146392/CE5_fire-tv-remote-cover-mb.zip)

LastDragon-ru, 2020-08-31:
This is a very old (and annoying) bug... As a workaround, you can try change Line width a bit: 0.4 =>0.399, 0.6 => 0.599, etc

[image]

Ghostkeeper, 2020-09-01:
Duplicate of https://github.com/Ultimaker/Cura/issues/7687 .

The problem is that you have two curves which are not actually curves. They are straight, flat surfaces with vertices and edges in between them. When there is a vertex on the inner curve but not on the outer curve, the model is slightly thinner there. When there is a vertex on the outer curve but not on the inner curve, the model is slightly wider there. As a result of bad luck, it manages to fit an extra wall in the wider parts but not in the thinner parts. See that thread for some workarounds and solutions that we're working on.

closed 2020-09-01 by Ghostkeeper