Ultimaker/Cura #8279: [4.7.0] Slicer adding extra geometry

bardakus, 2020-08-30:
**Application version**
4.7.0

**Platform**
Windows 10 Enterprise

**Printer**
Creality CR-10s pro

**Reproduction steps**
I loaded the model as is and sliced it. It was at this point that the box suddenly had a lot of stuff in the middle that it did not have before

**Actual results**
A box with unwanted floors in the middle

**Expected results**
A box without unwanted floors in the middle

**Project file**
[files.zip](https://github.com/Ultimaker/Cura/files/5147427/files.zip)

smartavionics, 2020-08-30:
The model's normals are out of whack (technical term), once fixed, it slices more like you would expect...

[image]

You can use the Cura Mesh Tools plugin to fix the normals.

bardakus, 2020-08-30:
Ok thanks for the help, I was not aware that would be an issue

closed 2020-08-30 by bardakus

Ghostkeeper, 2020-09-01:
Normal vectors affecting the print is an issue. They shouldn't. It's been known to us for a while though.