Ultimaker/Cura #8282: Backspace don´t delete object

gomakekit, 2020-08-31:
**Application version**
(The version of the application this issue occurs with.)
Cura 4.7

**Platform**
Mac OS Catalina 10.15.6
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Printer**

Ender 5
**Reproduction steps**

Import any object, press backspace, nothing happens. 

Additional info: There is no delete button on MacBookPro. Backspace should be the button for deleting objects.

**Screenshot(s)**

**Actual results**
Nothing happens

**Expected results**
Object should be deleted

**Project file**
Any object

**Log file**

**Additional information**

nallath, 2020-09-01:
The odd thing is that we have fixed this a few times over. The last time was in this issue: https://github.com/Ultimaker/Cura/issues/7754

Ghostkeeper, 2020-09-07:
Both of our QA engineers tried this on their Macs and couldn't reproduce the issue. Maybe it happens only on certain types of keyboards? What keyboard are you using?

Maybe we'll be able to tell more if there are more users with this problem.

gomakekit, 2020-09-21:
Hi. It´s the standard keyboard on a macbook pro 13-inch, 2017, macOS Catalina 10.15.6. I have now tested with the latest version of Cura and now the the delete works just fine! Thank you!

konskarm, 2020-09-21:
Perfect! I am closing this again then.

closed 2020-09-21 by konskarm