Ultimaker/Cura #8284: Cura with Flashforge Guider 2s may end in a broken leveling sensor.

oe3gwu, 2020-08-31:
**Application version**
4.6.2

**Platform**
Windows 10 Pro

**Printer**
Flashforge Guider 2s

**Reproduction steps**
I sliced a STL File. Cura began to print a purging line, waited for ooze, made a wipe (short up of the bed) and then before moving to the starting location, the leveling sensor was downed and it broke up the sensor, because it actually was in front of the bed, I know you do not support 3rd party printers, but because the starting gcode works on other platforms, I am not sure if it is a software issue.

**Screenshot(s)**
I dont have one. I wasnt prepared for this.

**Actual results**
I shut down the printer to avoid further complications.

**Expected results**
Move to starting location WITHOUT enabling the leveling sensor.

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
I am not sure if upload of the files worked properly. I added a download link too.
https://cloud.oe3xnr.eu/index.php/s/HVtiNAdth4CGSDV
[Cura Filament Sensor Issue.zip](https://github.com/Ultimaker/Cura/files/5152148/Cura.Filament.Sensor.Issue.zip)
[cura.log](https://github.com/Ultimaker/Cura/files/5152140/cura.log)
[image]

**Additional information**
I used starting gcode from Simplify3D. I use it since 1 year, no issues there.

nallath, 2020-09-01:
Oof :( That really sucks. 

Did you have time to look at the start g-code and see what is wrong with it? If you can provide us with a suggestion of what it should be, I'd be more than happy to make the change for you. You're right that we don't support it, but that doesn't mean we can't / won't help out fixing this. We just can't actually test it, since we don't have the machine.

oe3gwu, 2020-09-01:
I do have a suggestion.

When sending the command M402, the sensor is retracted, and this cant happen. Cura should somehow check that from beginning until the end, that the sensor is retracted.

Flashforge Guider 2 uses a Marlin Firmware Variant. I dont know the code to activate the sensor.

However, I dont thing that adding M402 to the start gcode would help, because this happens AFTER the purge line, that is printed correct with retracted sensor.

It must happen somewhere here.

```
G1 X120 Y-125 E20 F2000 ; extrude a line of filament across the front edge of the bed
G1 X130 Y-125 F180 ; wait for ooze
G1 X140 Y-125 F5000 ; fast wipe
G1 Z1 F100 ; lift
G92 E0 ; zero extruders again
```

This is my start gcode (copied from S3D - it is somewhat generic for all Flashforge printers):

```
M118 X10 Y10 Z10
M146 r255 g0 b0 ; set lights to red while preheating
M907 X100 Y100 Z40 A100 B100 ; set motor current
M132 X Y Z A B ; recall home offsets
G28 ; home axes
G1 Z100 F2400 ; lower Z
G161 X Y F4800 ; move to wait position
M7 ; wait for bed to stabilize
M6 T0 ; wait for right extruder to reach target temp
G161 X Y F6000 ; move to front left corner of bed
G1 Z0.3 F3600 ; move down to purge
G92 E0 ; zero extruders
G1 X120 Y-125 E20 F2000 ; extrude a line of filament across the front edge of the bed
G1 X130 Y-125 F180 ; wait for ooze
G1 X140 Y-125 F5000 ; fast wipe
G1 Z1 F100 ; lift
G92 E0 ; zero extruders again
M146 r255 g255 b255 ; turn lights on for print
G1 Z0.5 F3300 ; position nozzle
```

I already destroyed 1 sensor. I dont care to destroy another one or two, so I am your friendly beta tester for this issue.

nallath, 2020-09-01:
Could you try and run the start-gcode line by line (eg with something like printrun) to see in what line it's going wrong? 

It's a bit hard for me to suggest any changes without having access to the actual printer.

oe3gwu, 2020-09-01:
I can do. I bought 2 leveling sensors today. I am not sure when they are shipped. But I think I can try this also without. I never done this though. If I have got issues, I am going to post them here.

oe3gwu, 2020-09-07:
Update: I still havent got a new sensor. Because the Servo is also not installed, I cannot test the gcode. I will keep you updated.

oe3gwu, 2020-09-23:
Update: New Leveling Sensor shipped. Should get it at least next week. Then I can perform the tests.

oe3gwu, 2020-09-28:
Update, I got the new sensor today. Will do the tests on wednesday.

oe3gwu, 2020-09-30:
Ok, I tried to send the GCODE manually. I didnt find a software I could use with a guider 2s to send it line per line. Reason. The Guider 2s isnt recogniced as Serial Port. And Printrun (& co.) cant work with LAN Printers. Again the GCODE above works fine with S3D, but not with Cura.

Ghostkeeper, 2020-10-02:
So if I understand from your tests, you're sending the same exact g-code through a serial port with Cura and with Simplify, but the g-code gets interpreted by the printer differently when Cura sends it?

oe3gwu, 2020-10-02:
I am not sure if the start gcode is the issue. I just know, that my start gcode works in S3D, The SAME Code was used in Cura. Just I dont know if the start gcode was interpreted different, or something after the start gcode was sent to the printer that made this issue.

I used a USB Stick for printing. The issure is, that my Guider 2s doesnt emulate a Serial Port, so I cant run printrun.

Maybe it helps, if I slice something in S3D and in CURA and you compare the GCODE.

Ghostkeeper, 2020-10-05:
Debugging your specific printer is not something we can really help you with without access to the printer. And not on Ultimaker's time either. We'd like to be able to help, but to do that we need to directly know what needs to change for your printer.

It could be that there is a difference in the start of the print just before or just after the start g-code. Comparing the g-codes should make that visible.

UhohNoParentheses, 2022-01-23:
New to the Flashforge Guider II and Cura, I decided to tackle this. I noted same phenomenon. IE the bed height servo dropping down fully or sometimes partially just prior to extruding. I manually edited the gcode file to remove everything except the start and end code and noted no servo actuation. 

I then looked at the first line after the start code. M141, which is "Set Chamber Temperature" and commented it out and retried the gcode file. The printer then proceeded to initialized and start printing with no servo action. I checked the machine settings for my custom entry for the Guider II in Cura and noted the box was checked for "Heated Build Volume." After unchecking the box and saving another gcode slice, I noted the M141 command no longer appears in the gcode.

Ghostkeeper, 2022-02-01:
Thanks for the information! I hope that'll help with people looking for issues with the FlashForge Guider.

If you have a profile to add to Cura with working settings, we'd happily merge those into Cura. Until then though, I'll close this as simply being wrong settings, that enable g-code features that the printer doesn't handle well.

closed 2022-02-01 by Ghostkeeper