Ultimaker/Cura #8286: Open Project window self clearing without selection.

cbfasi, 2020-08-31:
**Application version**
$.7

**Platform**
Windows 7

**Printer**
Ender 3 Pro

**Reproduction steps**
1. Double clicked on multiple project files (.3mf's)(Something you did.)
2a. Cura opened, window with selection to open as project or model cleared by itself within seconds and not loading project in any form.
2b. Cura opened in background, slected tab with OPen as Project option and it self cleared without selecting and option and not loading project in any form

**Screenshot(s)**
Non available since it cleared!

**Actual results**
File is not loaded, this used to work ok before 4.7

**Expected results**
The Open project window should not of cleared and I should of been able to choose which load method I wanted (I dont always choose the same)

**Project file**
n/a

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
Have re-installed ensuring Cura set to load .3mf files, made no differance.
Only current way to load project is from within Cura, not from explorer etc.

cbfasi, 2020-08-31:
Sorry title should start [4.7]

nallath, 2020-09-01:
Please provide a log file and check if you didn't accidentally have the setting changed in preferences on what to do with projects.

cbfasi, 2020-09-01:
Preferences checked, all ok.
The fact that I can still select the choices when opening a project from within Cura also proves settings are correct, the problem is how Cura reacts when a file is selected for opening outside of Cura.

Replicated problem a number of times this morning too, have attached
 log from the time I ran the tests (not attached full log from last night)

[Problem cura.log](https://github.com/Ultimaker/Cura/files/5155188/Problem.cura.log)
[image]

Ghostkeeper, 2020-09-02:
Could you share one of the 3MF files you're trying to open? From the log it looks like you're opening a plain 3MF file, not a Cura project file. Maybe the file was created using Cura's File -> Export..., or created in a different application than Cura.

cbfasi, 2020-09-02:
File created using File -> Save Project... in Cura.
All my 3mf's are, I have no other way of making them, and tbh I only use them save Cura Project files (better than saving just g-codes, can go back and change settings if project!)
[Northampton 1-3000.zip](https://github.com/Ultimaker/Cura/files/5162861/Northampton.1-3000.zip)
3mf in zip as Github wont accept 3mf files by themselves.

cbfasi, 2020-09-02:
If you open from the above from within Cura all works fine....
BUT when double clicking the file from outside Cura, Cura starts up, briefly shows the Open as Project or Model window then self closes it without any selection being made.  
This is significantly slowing the use of multiple project files as I have to be in Cura to open it, and not from outside Cura (even though its clear Cura knows what to do, just self clears )

Ghostkeeper, 2020-09-02:
Hmm I do get the pop-up (and it stays) even when I open the 3MF file with Cura.
[image]

Maybe that's something happening in Windows only? Is some developer on Windows able to reproduce this issue?

cbfasi, 2020-09-02:
I dont even see that screen if opened from outside of Cura

cbfasi, 2020-09-02:
Just renamed the log file to create a new one for when I double click on the above project file, now saved off log file before carrying out any action within cura.  attached below, this will of removed any 'extra' log information
[Northampton-Cura.log](https://github.com/Ultimaker/Cura/files/5163200/Northampton-Cura.log)

Ghostkeeper, 2020-09-03:
The log doesn't really show anything there, it's as if the file was never loaded.

We haven't been able to reproduce this, so far. When the preference is set to "Import models" then yes, but you showed that was not what the preference is set to :( I don't know how we can debug this at the moment.

rburema, 2020-09-08:
Can't reproduce this either, the file you provided opens just fine on Win64 in 4.7 for me.

All I can say is that I wonder if it is in some way related with either https://github.com/Ultimaker/Cura/issues/8147 or https://github.com/Ultimaker/Cura/issues/6919

That said, sadly, we'll have to be able to reproduce the issue (or, failing that, some log message showing up) if we have any hope of an attempt at a fix :-(

cbfasi, 2020-09-12:
Upgraded to 4.7.1, seems its fixed, I think I gave Cura a heart-attack when I told it to open 20 3mfs at same time (I allow multiple Cura instances).. and each one waited for me choose ! (had already checked with a couple and thought lets really check)

We may never know exactly what was causing it but whatever it was, is not currently an issue.

Thank you for trying to fix it intentionally!

closed 2020-09-12 by cbfasi