Ultimaker/Cura #8287: 4.7.0 - Imported Profiles Do Not Show in List

ToaKamate, 2020-09-01:
**Application version**
4.7.0

**Platform**
Window 10 Pro 64-bit

**Printer**
Ender 3

**Reproduction steps**
Installed 4.7.0.  Normally an upgrade picks up all previous custom profiles. I noticed one was missing.
I imported the profile via manage profiles.  Cura indicated it was imported (tried several times).
The imported profile doesn't show up under the customer profiles.
Renamed .curaprofile file and imported again, twice - still does not show.
Installed 4.6.0 and it picked up the original TPU profile from the 4.5 install.

**Screenshot(s)**
See attached.
[image]
[image]

**Actual results**
imported profile does not show.

**Expected results**
imported profile should have been added to list of custom profiles

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
see attached
[cura.log](https://github.com/Ultimaker/Cura/files/5153193/cura.log)

**Additional information**
(Extra information relevant to the issue.)

ToaKamate, 2020-09-01:
Update on the above.  I have tried to import several other profiles with the same result.  Cura says import was successful, but the profiles do not show in the list of custom profiles.

nallath, 2020-09-01:
Could you also share the curaprofile that you tried to import?

ToaKamate, 2020-09-01:
I tried. Forum does not allow that file type. I'll post somewhere and provide the link after my coffee kicks in.

ToaKamate, 2020-09-01:
OK, zip attached.  1 Minute mp4 file shows what is happening.  Interesting, the Magic0.12 profile imported and shows in list.  The TPU Flex and Ninjaflex profiles import - but don't show in list.  

However, I can successfully import the TPU and Ninja profiles into 4.4, 4.5 and 4.6 without issue.
[CURA Profiles.zip](https://github.com/Ultimaker/Cura/files/5157492/CURA.Profiles.zip)

nallath, 2020-09-02:
Ah, I see already. This is a duplicate of https://github.com/Ultimaker/Cura/issues/8229

closed 2020-09-02 by nallath