Ultimaker/Cura #8288: v4.7 fails to adjust settings from .3mf file

KenLucke, 2020-09-01:
Loading a project file (.3mf) no longer sets slicing parameters that were in place when the file was saved, as it used to up to v4.6

**Application version**
4.7

**Platform**
Mac OSX 10.15.6

**Printer**
CR-10s

**Reproduction steps**
1. load .3mf file
2. settings do not change to match from when file was saved

**Screenshot(s)**

**Actual results**
Nothing...  I have to reset all settings if they have been changed

**Expected results**
.3mf files used to change parameters to what they were when file was saved

**Project file**

**Log file**

**Additional information**

nallath, 2020-09-01:
Please provide the 3MF file that has this issue. Please also provide a log file spanning the period where this happened

KenLucke, 2020-09-03:
You may disregard.  After much digging and experimenting, I found that apparently, when I upgraded, it somehow reset the General setting (I know that I did not do it manually) on what to do when opening a Project file to "Always Import Models" instead of  "Always Open as a Project."

I will close this issue now.

closed 2020-09-03 by KenLucke