Ultimaker/Cura #8292: Better settings navigation

gmarkowitz82, 2020-09-01:
**Is your feature request related to a problem? Please describe.**
1. I'm always frustrated when I "right click" the Material Section under "Prepare" and do not get an option to jump directly to the material settings
2. I would like to be able to set the "Flow" of my material within the Material's Print Settings, so that gets recalled within profile configurations
3. 1. I'm always frustrated when I "right click" the Printer  under "Prepare" and do not get an option to jump directly to the printer settings
4. I am always aggravated when I'm working on the above listed sections, and I scroll over a value, while my mouse is over it, and it changes all my values and I cant remember what they were because I wasn't looking for them in the first place. 

**Describe the solution you'd like**
1. Please add the features listed above
2. I would like to be able to "Save" settings within the above areas, to lock them in

**Describe alternatives you've considered**
1. With the scroll wheel, just maybe sub out the box type so you cant scroll through the values

**Affected users and/or printers**
EVERYONE

**Additional context**
If these are in the next version, Ill bake you all cookies

nallath, 2020-09-01:
>     1. I'm always frustrated when I "right click" the Material Section under "Prepare" and do not get an option to jump directly to the material settings

Type in materials in search or use the custom visibility plugin to create a "materials" visability

>     2. I would like to be able to set the "Flow" of my material within the Material's Print Settings, so that gets recalled within profile configurations

Check the material settings plugin.

>     3. 1. I'm always frustrated when I "right click" the Printer  under "Prepare" and do not get an option to jump directly to the printer settings

I don't know what you mean by that. 

>     4. I am always aggravated when I'm working on the above listed sections, and I scroll over a value, while my mouse is over it, and it changes all my values and I cant remember what they were because I wasn't looking for them in the first place.

I thought that was something we fixed some time ago.

gmarkowitz82, 2020-09-01:
Please see attached photos as a visual aid to my presentation
[image]
[image]

nallath, 2020-09-01:
As for the quick opening of material settings, use ctrl + k

Ghostkeeper, 2020-09-02:
> 2. I would like to be able to set the "Flow" of my material within the Material's Print Settings, so that gets recalled within profile configurations

Duplicate of https://github.com/Ultimaker/Cura/issues/2058.

It would help for the future to limit issues to one issue per thread, to prevent confusion as to whether the issue is solved yet or not.

Ghostkeeper, 2020-09-02:
> 4. I am always aggravated when I'm working on the above listed sections, and I scroll over a value, while my mouse is over it, and it changes all my values and I cant remember what they were because I wasn't looking for them in the first place.

Your operating system creates these widgets. Perhaps your operating system has a way to customise numerical text areas? We've had a plan long ago to replace the entire preferences window with custom elements in order to style/theme them properly. But we haven't gotten around to that yet; the idea is never too pressing.

closed 2020-09-02 by Ghostkeeper