Ultimaker/Cura #8293: anti Layer Shift option: home horizontal axes checkpoints

daT4v1s, 2020-09-01:
**Is your feature request related to a problem? Please describe.**
occasional layer-shifts on high jerk values for certain specific prints & locations 

**Describe the solution you'd like/propose & why**
add option for X& Y axis homing 
- between layers
- after long print distance
- between line types (wall, top/bottom, infill)

if there is a layer shift, then it's impact would be limited to a layer or less and much easier 
to minimize the loss/defect, easier to post-process

**Describe alternatives you've considered**
a simple alternative is to add gcode in between layers via plugin
```gcode
G91 ;relative positioning
G0 Z+0.4;
G90 ;absolute positioning
G28 X;X Home
G28 Y;Y Home
G91 ;relative positioning
G0 Z-0.4;
G90 ;absolute positioning
```

also this does not allow for inserting homing between say top-bottom-shell and  wall
line print distance + discontinuous geometry (e..g complex top/bottom-shell+ multiple separate walls) seems to increase risk of layer shifts, 
maybe using a weighted score to opportunistically insert homing

the travel move to the XY home is not optimized,
g-code use of homing the 2 axes in attempt to avoid travelling over printed parts
is not as good as tailed travel moves,
it requires a retract before the home, lest ooze-stringing occurs,
maybe not familiar with g-code enough yet, currently requiring a z-hop

maybe certain inter-layer travel move slightly affect risk also

if homing tendency is skewed more for layers  with more print-length
perhaps it would distribute  risk vs time-cost  more evenly

**Affected users and/or printers**
prints with layer-shift-risk, to minimize material loss

**Additional context**
(Add any other context or screenshots about the feature request here.)
this may require knowledge info about printer
where the X/Y axis home is (what combination of X/Y  minimum or max)
and unsure how would apply to other printers (non-cartesian, CoreXY, delta, radial)

Ghostkeeper, 2020-09-03:
One problem with homing during the print is that it tends to destroy previously printed objects when printing in one-at-a-time mode.

I don't know if that's what you meant, but you can add that g-code between every layer using the "Insert at layer change" post-processing script in Cura 4.7. You can add a retraction there too. So the things that are not possible then are:
* Triggering a homing at certain salient points in the print, e.g. when switching between structures. There is no inherent difference between the structures that would cause a layer shift though. It only prints at a different print speed, line width, etc. But maybe what you're seeing is that the travel move tends to be a bit longer there.
* Combing. You'll have to use a Z hop instead.
* Maybe having a good user experience, because writing g-code in a post-processing script is something only experts will do.

However in my opinion, when you have a layer shift it's already too late. I'd take a good hard look at the travel speed/acceleration/jerk, because reading your feature request, I think that's where your problem actually lies.

Ghostkeeper, 2020-09-03:
You could also do a Search and Replace post-processing script that replaces stuff like `;TYPE:INFILL` by `;TYPE:INFILL\nG28 X Y` or something, to trigger a home every time it starts with infill.