Ultimaker/Cura #8294: Cura 4.7.0 no shared heater checkbox

B-rad888, 2020-09-01:
**Application version**
4.7.0 

**Platform**
Windows 10

**Printer**
Custom fff printer

**Reproduction steps**
1. Add custom FFF printer
2. Select 2 nozzles
3. No option for shared heaters

**Screenshot(s)**
[image]

**Actual results**
no checkbox

**Expected results**
checkbox

**Project file**
on github

**Log file**
no log

**Additional information**
Not mentioned in changelog. Maybe missed in the update?

smartavionics, 2020-09-01:
See https://github.com/Ultimaker/Cura/issues/8148

Ghostkeeper, 2020-09-03:
Let's keep the discussion in #8148 where this was decided.