Ultimaker/Cura #8298: Cura 4.7 clicking custom profile does not update settings list in manage profiles window.

brunoosti, 2020-09-02:
The list on the right side of the window does not update with my custom profiles when 
I click on them. It shows the "normal" pre configured profile settings/numbers.
(Windows 10, cura 4.7, anycubic i3 mega-s)

nallath, 2020-09-02:
Please fill in the entire bug template, preferably with screenshots of what is happening and a complete description of what you think should happen

brunoosti, 2020-09-02:
Sorry. I'll try to explain it better below.
See on the picture I have my "PLA - MINHA CONFIG" selected both in the manager and in the main slicing window (with no modification to it). Notice that Global Settings (profiles manager) are not the same values as Print settings on the main window (infill density, for an instance) even though the manager shows the message "your current settings match the selected profile". Those setting are from the "normal" default profile. Is this the correct way of this feature to work? (I can't remember if it was like this before). If it is, what is the logic behind it? Anyone would imagine that the list would show the selected profile settings, wouldn't it? (I'm confused) Thanks : )
[image]

closed 2020-09-03 by Ghostkeeper