Ultimaker/Cura #8299: [Material Type]  -- Change to Drop Down List

3Deee, 2020-09-02:
**Application version**
4.7

**Platform**
Mac IOS

**Printer**
Ender 5

**Reproduction steps**
From the Properties menu >> Material  > [Material Type] is an editable, type-in entry field. Print Profiles are linked to this field. Profiles appear when [Material Type] is recognized. Profiles do not appear when [Material Type] is not recognized. Examples: When "PLA" is entered, all Profiles appear. When something else is entered, say "PLA Performance" which has slightly different material properties, Profiles do not appear. PLA Performance is a Keene Village filament. Changing [Material Type] field to a pull down list of acceptable entries would alleviate issue.

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(see above) 

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

3Deee, 2020-09-02:
I came across problem and resolved it before reading other strings. Others suggested same solution. They must have a great mind too. (just kidding).

Ghostkeeper, 2020-09-03:
We already have this on our backlog, arising from this issue: https://github.com/Ultimaker/Cura/issues/8229

Good to have a separate ticket about it I guess. But if we fix this then we should also consider the original fixed.

RaabenF, 2020-09-03:
thank you for clarification.