Ultimaker/Cura #8300: [4.6] Since Cura ~4.4 slicing not possible anymore (also 4.7)

maxpd1, 2020-09-02:
**Application version**
Mainly 4.6. But also 4.7 on one PC

**Platform**
Notebook and Desktop Windows 10.

**Printer**
Anycubic i3 Mega

**Reproduction steps**
1. I start cura
2. I import a model
3. I press Slice

**Screenshot(s)**
Here is where it stops:
[image]

**Actual results**
Cura doesn't slice

**Expected results**
Cura should start slicing

**Project file**
Every project file I tried during the last half year.

**Log file**
Complete Log of 4.6 attached

**Additional information**
With Cura 4.7. on my Notebook slicing is possible if I turn of the firewall. Even if all Cura interfaces are allowed to communicate through firewall. 
With Cura 4.6 is isn't even working with deactivated firewall.

[cura.log](https://github.com/Ultimaker/Cura/files/5160221/cura.log)

nallath, 2020-09-02:
As far as I can see from the logs, the slicing happened as expected.

maxpd1, 2020-09-02:
Puhhh, that's even more strange :D

Ghostkeeper, 2020-09-03:
Indeed in the log it says that it took 14.8s to slice. You then saved the g-code to `D:/AI3M_Lampenfuss6.gcode`. Is that file empty or something? What happened with that file?

closed 2020-09-29 by no-response[bot]