Ultimaker/Cura #8303: Importing profile successful but profile not visible

VapeN3D, 2020-09-02:
**Application version**
4.7

**Platform**
Windows 10, Nvida GTX, 16gb RAM
**Printer**
Ender 2

**Reproduction steps*

**Screenshot(s)**

**Actual results**
No profile visible 

**Expected results**
Profile should be visible and useable 

**Project file**

**Log file**

**Additional information**
(Extra information relevant to the issue.)

VapeN3D, 2020-09-02:
[cura.log](https://github.com/Ultimaker/Cura/files/5163088/cura.log)
[image]

VapeN3D, 2020-09-03:
Rolled back to the previous version of Cura and now I can see the Ninjaflex profile

Ghostkeeper, 2020-09-03:
Duplicate of https://github.com/Ultimaker/Cura/issues/8260 and https://github.com/Ultimaker/Cura/issues/8229 .

We're planning to change the wording of this dialogue if the imported profile is not applicable to the current configuration.

closed 2020-09-03 by Ghostkeeper