Ultimaker/Cura #8305: Cura pause at layer height pauses at wrong layer

martijnschouten, 2020-09-02:
**Application version**
Cura 4.7.0

**Platform**
Windows 

**Printer**
Diabase H-series

**Reproduction steps**
1. Slice my model with pause at height at 0.4 mm with 0.1mm layer height and 0.1mm initial layer height.
2. Save the gcode.

**Screenshot(s)**
[image]

**Actual results**
The gcode includes a pause at the start of the 0.3mm layer and moves back to a height of 0,4mm

**Expected results**
The gcode includes a pause at the start of the 0.4mm layer and moves back to a height of 0,4mm

**Project file**
[model.zip](https://github.com/Ultimaker/Cura/files/5164734/model.zip)

**Log file**
[cura.zip](https://github.com/Ultimaker/Cura/files/5164749/cura.zip)

rburema, 2020-09-04:
Confirmed. It seems the 'current z/height' is set incorrectly during in the pause at height script at this point for some reason.

Ghostkeeper, 2020-09-04:
Duplicate of https://github.com/Ultimaker/Cura/issues/5085. It's a bit of a weird issue, this one. Happens only with certain numbers apparently.

closed 2020-09-04 by Ghostkeeper