Ultimaker/Cura #8306: Changing nozzle size doesn't update line width

dxgldotorg, 2020-09-02:
**Application version**
4.7

**Platform**
Windows x64, GTX1080

**Printer**
HyperCube (custom profile)

**Reproduction steps**
Change nozzle size in Machine Settings

**Actual results**
Line width is not updated even when set to default

**Expected results**
Either update the line width in print settings or display the revert button

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Additional information**
I often swap out nozzle sizes on my 3D printer and having to manually change then reset line widths is a bit annoying.

Asterchades, 2020-09-03:
This is one of those situations where including the profile would help. If it's got a fixed or restricted line width than it's a localised problem you'll need to alter the profile to address.

I know I use a completely custom profile, but I have a line which specifically reads _"line_width": { "value": "machine_nozzle_size" },_ to compensate for things like that automatically.

dxgldotorg, 2020-09-03:
It happens when my line widths are at default aka the size of the machine nozzle.  The values don't update automatically and I have to set some bogus size then revert to pull the correct size.

rburema, 2020-09-04:
With a custom-fff-printer, if I change the nozzle size from 4 to 3, it didn't update initially, but does so after I switch profiles back and forth. So this is confirmed then. I'll bring this up during the discussions.

closed 2020-09-07 by nallath

Ghostkeeper, 2020-09-07:
Fixed for 4.8.

commit 349f3b7e (2020-09-07): Force update when changing nozzle size in custom FFF

Fixes #8306