Ultimaker/Cura #8310: Configuration file problem (4.6.2 => 4.7.0)

Fabdu57, 2020-09-03:
Hello,

I have a problem with one of my configuration file.
On 4.6.2, it works.
On 4.7, when i try to import, i have a message that say configuration correcly imported, but it don't appear on list. If i try again, i have same message but with #2, but always nothing on list...

How to solve this bug ?...

Thanks in advance,
Fabrice
[Cura v2020.06.28 - Litophanie.zip](https://github.com/Ultimaker/Cura/files/5169334/Cura.v2020.06.28.-.Litophanie.zip)

rachael7, 2020-09-03:
I had this problem with one of my profiles. It was importing it, but not making it available for the machine I was on, or maybe any machine at all. It did the same thing with adding copy numbers to the end. Eventually, I was able to track down the problem by comparing the profile to the other profiles that did work. There was one line that was different, I think it was 'quality_type'. In the working profiles, it was set to 'not_supported', but in the bad one, it was set to something else. To get at it, I had to open the .curaprofile file with a zip extractor, change the problem line, then put it back in the zip file. After I did that, it imported fine. It was still #5 or something like that, but I went in the folder C:\Users\my_user_name\AppData\Roaming\cura\4.7\quality_changes, deleted #1-4, then I was able to rename the one that worked in Cura.

Fabdu57, 2020-09-04:
Thanks for your message.
Effectively i found on "C:\Users\$user\AppData\Roaming\cura\4.7\quality_changes" the cfg file associate to my configuration:

creality_base_extruder_0_cura_v2020.06.28_-_litophanie.inst.cfg
creality_ender3pro_cura_v2020.06.28_-_litophanie.inst.cfg

I tryes to delet them and tooh from 4.6 (C:\Users\$user\AppData\Roaming\cura\4.7\quality_changes) both and put on 4.7, but still same problem... Then i tryes a second solution : import as you mention and rename file. Same, profile not appears on list.

I think that 4.7 version have a lot of bug... And that is a big bug that Ultimaker need to interven ASAPS...

rburema, 2020-09-04:
> Attached is the profile.

I don't see it here. Could you try that again? It may need zipping to appease github.

Fabdu57, 2020-09-04:
[Cura v2020.06.28 - Litophanie.zip](https://github.com/Ultimaker/Cura/files/5173072/Cura.v2020.06.28.-.Litophanie.zip)

nallath, 2020-09-04:
Duplicate of https://github.com/Ultimaker/Cura/issues/8229

closed 2020-09-04 by nallath

Fabdu57, 2020-09-06:
> Duplicate of #8229

Sorry but ly problem is not solv...
Is a solution avaible or not ?!

rburema, 2020-09-06:
@Fabdu57  No sorry, not yet. This one is closed because we're sure it's the same as #8229 (which is still open), and we like to keep issues that where already reported in the same place. To continue to conversation and/or to monitor the discussion/progress, you should go to that link, which goes to an open issue that is the same (or has the same underlying problem) as yours.

Fabdu57, 2020-09-11:
This is not my problem that an other topic is open.
My problem is different, i am not the other people.

So i want my topic became again open and have help !

konskarm, 2020-09-11:
Hi @Fabdu57. The profile that you are trying to import is a draft profile which is not compatible with nozzlies <0.6mm. The quality profile is correctly being imported, but it does not become available due to your selected nozzle. If you switch to a nozzle that contains a draft profile (e.g. 0.6mm), you should be able to see the quality profile you just imported. 

I agree that currently it is not clear, so we are working on a solution to at least give a message as to why the profile is not visible. 

The reason this github issue is closed is that it is a duplicate of https://github.com/Ultimaker/Cura/issues/8229 and we want to keep the discussion on duplicate issues in one centralized issue, so please move further comments on https://github.com/Ultimaker/Cura/issues/8229.

Fabdu57, 2020-09-12:
I don't reallly understand...
It works on 4.6.2, layer height is 0.12 and noozle 0.4.
So i don't really understand your answer in fact.... If it works on 4.6.2 and i can print, why not on 4.7 lol

konskarm, 2020-09-14:
In 4.6 it was actually a bug that was showing all of the quality profiles, even if they were not compatible. We fixed that in 4.7 so as not to allow a configuration select a profile that doesn't match it, which can cause bad or even failed prints.