Ultimaker/Cura #8312: [Feature Request] Selectable profile settings changes

KenLucke, 2020-09-03:
Please add checkboxes to enable/disable individual changes to settings when changing a profile.  There are many times I want to change to, say, a different material, printer, etc., and when I do so, I have only the choices to keep or discard all differences, rather than keeping some and discarding some.

I picture a checkbox next to each setting that would be different (the ones that show up in that dialog now), and the "keep" or "discard" button* would affect only those marked, plus a "select/deselect all" button would be very useful.

*i.e., if you click "Keep", it would keep only those that were marked, but if you clicked "Discard," it would only DISCARD those marked.  An alternative is to only have one button ("Accept"), and resulting only keeping those marked, discarding the others.

rburema, 2020-09-04:
Keep (pun not intended) in mind that people are already grilling us that that particular screen is too complicated:
  https://github.com/Ultimaker/Cura/issues/7592
  https://community.ultimaker.com/topic/33410-discard-or-keep-changes/?tab=comments#comment-270035 (Note: The OP of this second link gets a bit ... unpolite about their grievances, not just at us but at other users trying to help them. I put this remark here to show to anyone that might be reading that this isn't in general acceptable.)

I'll bring this up during discussions, but it seems there might be some hidden complexity here as well, so it might get deferred.

KenLucke, 2020-09-04:
> Keep (pun not intended) in mind that people are already grilling us that that particular screen is too complicated:

LOL, as opposed to the program in its entirety?  :)   I can understand it adds some complexity, both to that dialog and to the programming, but if someone can handle the immense numbers of other things that have to be specified and dealt with in the program as a whole when doing any sort of serious tweaking of your model during a slice setup, they should be able to handle that dialog, especially if would default to exactly the same as it is now (all check boxes marked, "Keep" or "Discard" buttons active ,so all you have to do is click the same button you do currently).  

I'm always in favor of giving the user ultimate end control over their program, and was always taught in my programming classes to give at least two ways to do the same thing, if possible.  Sadly, my programming days were back in the COBOL, Fortran, Pascal, Assembly, C, C++, and Machine Language days [including clear back to the days of flipping a bank of 8 switches to set code and registers], and, alas, I can't keep up any more.

Maybe for the whiners, one MORE preference (I know, I know <grin>) , this time back in the General Settings, an "Advanced Profile Change Dialog" option that would toggle between the simple current one and the one I proposed. :)

I realize it may never be done, it's just a pie in the sky wish.  I have a lot of those. :)

Ghostkeeper, 2020-09-07:
> I'm always in favor of giving the user ultimate end control over their program, and was always taught in my programming classes to give at least two ways to do the same thing, if possible.

There is a delicate balance to this though. If the interface looks too much like a cockpit, it scares off novice users and confuses advanced users. Only the most expert users (which also happen to be the type that reports things here on Github) will understand the interface and only a fraction of those actually use that functionality. So it can do harm to allow changing every little thing in multiple ways.

Currently there is one way to do this (not multiple): You can press "keep" and in the next screen you'll be presented with a list of settings with revert arrows also.

Because the workaround is really just one click away, and because we need to be careful not to make the interface too complex (which would require some UX work to get right), we've decided to defer this issue. We realise it's something you might want to do, but the cost of implementing it is too high for what it's worth so it will not get any priority until things change.