Ultimaker/Cura #8313: [4.7]bugged 0g slices ender 3 pro

Dishyboar, 2020-09-03:
**Application version**
4.7

**Platform**
win 10 
ryzen 3600
rx 570
**Printer**
ender 3 pro

**Reproduction steps**
when i slice the preview menu is showing some weirdness and says 0g for the print amount. ive tried default profiles and

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 
[image]

**Project file**

[CE3PRO_Fighter01_Horns.zip](https://github.com/Ultimaker/Cura/files/5171662/CE3PRO_Fighter01_Horns.zip)

rburema, 2020-09-04:
That model isn't manifold ('watertight') and therefore may have those problems with slicing. In fact, it's broken enough that even the meshfix plugin couldn't fix it completely.

You may be able to see it in preview (the non-yellow parts that show are where the models is 'broken').

I've tried it with the newest version (4.7.0) as well as an older version (4.0.0) and I get basically the same result. Are you sure this was the version of that model that you printed before?

nallath, 2020-09-04:
There are actually _so many_ overlapping and self-intersecting faces, that the highlighting algorithm is having trouble displaying them correctly. 

I've loaded your model into meshlab and selected the self intersecting faces. There are ~71.000 incorrect faces in this model (which is almost 50% of the model). 
[image]

By enabling "Remove all holes" in the Mesh fixes category I could get it to print with some semblance of the actual model, but i fear that this model is just too broken for any real attempt to print it.

closed 2020-09-04 by nallath

Dishyboar, 2020-09-04:
i must have gotten lucky when i got it to print the first time, because repairing this model took me a long time. i have the g code cura 4.7 
[d9.zip](https://github.com/Ultimaker/Cura/files/5175697/d9.zip)
managed to produce if your interested.
(note that the print quality of this gcode wasnt great as you could imagine)