Ultimaker/Cura #8314: Printing Stops Hours into Print Job

gslindstrom, 2020-09-04:
**Application version**
Cura v 4.7.0

**Platform**
Windows 10 v1909, Build 18363.10.16

**Printer**
Ender 3  v1.1.6

**Reproduction steps**
1. Started a large print job (15h 48m) via cable to the printer
2. The print job ran like a champ for roughly 12 hours and then froze. The nozzle was still in contact with the item being printed. I spoke with a friend and they experienced a similar stoppage on a long job (not the same part or project).

**Screenshot(s)**

**Actual results**
The printing "froze" after (about) 12 hours of printing.

**Expected results**
I was expecting the print to run for about 5 more hours.

**Project file**

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

rburema, 2020-09-04:
The gcode that gets sent via cable should be the same as if it was printed in any other way.

To be honest there are a lot of external factors when printing via cable that are outside of Cura's control. If I could have the gcode produced and the layer at which it went wrong I could be a bit more confident that it isn't Cura though.

rburema, 2020-09-04:
Oh, and the logs, please.

eskeyaar, 2020-09-04:
Also what host you were using to send print job? Repetier host? Pronterface ?

gslindstrom, 2020-09-05:
Thank-you for your quick response! I use the cable connecting Cura to my
Ender 3. I kicked off another very large job which has run almost 18 hours
without a problem. I think my issue may be with the file or, perhaps, the
power "blinked". At any rate, it does not appear to be a problem with your
software.

--greg

rburema, 2020-09-06:
> At any rate, it does not appear to be a problem with your
software.

Thanks for reporting back. I guess we can close this issue then :-)

closed 2020-09-06 by rburema