Ultimaker/Cura #8315: [4.7.0] curaengine.exe Crashes

rachael7, 2020-09-04:
**Application version**
4.7.0

**Platform**
Windows 8.1 Pro, Version 6.3.9600 Build 9600, NVIDIA Quadro 7000

**Printer**
Ultimaker S5

**Reproduction steps**
1. Open Cura 4.7.0
2. Load .3mf project file
3. Attempt to slice

**Screenshot(s)**
[image]

**Actual results**
curaengine.exe crashes and slicing progress bar returns to zero after acknowledging Windows message

**Expected results**
Slicing should have completed, as it does with the same files in Cura 4.6.1

**Project file**
https://drive.google.com/file/d/17Pz6UtQ8FCmKtoY5nQNWBoQ2bGRELZmx/view?usp=sharing

**Log file**
[cura.log](https://github.com/Ultimaker/Cura/files/5172624/cura.log)
[DxDiag.txt](https://github.com/Ultimaker/Cura/files/5172626/DxDiag.txt)

**Profile file**
[UMS5, Fine, Custom.zip](https://github.com/Ultimaker/Cura/files/5172664/UMS5.Fine.Custom.zip)

**Additional information**
- This also happens with standard profiles, not just the custom profile I am using here. 
- If I open the same project file with the same profile in 4.6.1, it will slice normally. 
- In 4.7.0, sometimes if I start with a fresh build plate, place the models, set the profile, and slice, it will work. But if I then save that project file, close and restart Cura, and open the project file, it goes back to crashing curaengine.exe

rburema, 2020-09-04:
I've sliced this correctly in 4.7.0 twice now. Once with the original downloaded file, and once with a 'resaved' file like you mentioned in the additional information.

Given the size/intricacy of the model, it might be a memory problem? Can you open the memory usage tab in the task manager while (or before) this begins slicing and see how much it takes?

rburema, 2020-09-04:
Update: This might be related to a parallelisation bug that we just found. A workaround might be to set the stair-stepping to a single layer height.

rachael7, 2020-09-04:
Thank you for your efforts. Coming back to it this morning, the first thing I tried was a reboot of the computer. After the reboot, it sliced okay a few times, but then began having trouble again when I increased the complexity of the slice by adding another model to the build plate and going to a finer resolution. During slicing, cura.exe is using about 2.4-2.6GB.  Curaengine.exe peaked as high as 1.2GB early on, but then rapidly fell back to around 400MB and slowly increased back to about 800MB and held there for a while +/- 50MB. Just before crashing memory usage on curaengine.exe started rising rapidly, getting up to about 900MB before it crashed.  The machine has 32GB installed and is showing over 10GB free, but will free more as needed by releasing RAM disk commits.  After the crash, and per your suggestion, I set the "Support Stair Step Height" equal to the layer height, with no other changes, and tried again. It did successfully slice with that change, with memory using peaking at about 3.5GB for cura.exe and 1.5GB for curaengine.exe. Thanks again for your help!

Ghostkeeper, 2020-09-07:
As a workaround, try setting the Support Stair Step Height setting to 0.1mm (or more).

I'm also unable to reproduce the problem, but I agree that https://github.com/Ultimaker/CuraEngine/pull/1317 will probably fix this.

rachael7, 2020-09-07:
Thank you. I can confirm that this workaround is effective. I am able to slice without crashes, as long as Support Stair Step Height is set equal to the layer height. 

> As a workaround, try setting the Support Stair Step Height setting to 0.1mm (or more).
> 
> I'm also unable to reproduce the problem, but I agree that [Ultimaker/CuraEngine#1317](https://github.com/Ultimaker/CuraEngine/pull/1317) will probably fix this.

Ghostkeeper, 2020-09-08:
The fact that the workaround works makes it very likely that this is the bug that was fixed by Ultimaker/CuraEngine#1317. Now that that one is merged, I'll mark this issue as fixed. Thanks for the report! It's an important bug.

closed 2020-09-08 by Ghostkeeper