Ultimaker/Cura #8316: Hole Horizontal Expansion causes 4 layers to print instead of 99 - 4.7.0 beta

95gtspeedemon, 2020-09-04:
**Application version**
(The version of the application this issue occurs with.)
4.7.0

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)
win 10
**Printer**
(Which printer was selected in Cura?)
ender 3 pro

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)
changed hole horizontal expansion from 0 to .2, then to .5
finally figured out what caused the print to stop early.
setting it to .39 is ok, prints the hollow test cube with a 5mm hole in the bottom just fine. setting to .4 or higher makes it bug out.

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

95gtspeedemon, 2020-09-04:
[image]
[image]

rburema, 2020-09-04:
Is this a box with an open top? If so, it registers the inside of the box as a hole. It'll then expand the hole beyond the confines of the outer wall.

If that's the case: What you could do is apply a modifier mesh to the bottom hole and put the altered hole expansion parameter in there only.

If something else is going on, then you're really going to have to give us the saved project file, so it can be investigated.

95gtspeedemon, 2020-09-05:
[Hole Hollow Calibration Cube.zip](https://github.com/Ultimaker/Cura/files/5178374/Hole.Hollow.Calibration.Cube.zip)
[ASA Hole Hollow Calibration Cube .4 bug.zip](https://github.com/Ultimaker/Cura/files/5178386/ASA.Hole.Hollow.Calibration.Cube.4.bug.zip)

95gtspeedemon, 2020-09-05:
see if the gcode and .stl help

bremco, 2020-09-06:
Perhaps the setting 'Print Thin Walls' will help.

Otherwise, this problem might be fixed by the method I layed out above (only expand the bottom circular hole by applying the hole expansion only on a modifier mesh located there).

As for why this happens, and is, in fact, expected behavior, read on:

The inside of the box itself is also a hole. That (square) hole will also be expanded, just like the smaller circular one at the bottom. The outside however, will stay the same dimensions. At some point, the expansion will be large enough that the hole is close enough to the outside, so that no walls are generated.

[image]

Ghostkeeper, 2020-09-07:
Indeed, this is expected behaviour.

You have a mesh that consists of a hole with a 0.8mm wide rim around it. If you set the Hole Horizontal Expansion to 0.4mm or more, the rim will be reduced to 0.4mm or less. As a result, with a line width of 0.4mm, anything less than 0.4mm will not get printed. Print Thin Walls should be able to help or else to reduce the line width.

closed 2020-09-07 by Ghostkeeper