Ultimaker/Cura #8317: [4.7.0] Can not modify 'Build Plate Adhesion Type' for overlap

ncheltsov, 2020-09-04:
**Application version**
4.7.0

**Platform**
Windows 10

**Printer**
Creality Ender 3 V2

**Reproduction steps**''''
1. Create overlap region
2. Modify settings for overlaps
3. Select settings
4. Select Show all checkbox
5. 'Build plate adhesion' group is shown, but it has no elements when it is expanded.

**Screenshot(s)**

[image]

[image]

**Actual results**
'Build plate adhesion' group is shown, but it has no elements when it is expanded.

**Expected results**
I can create, for example, Brim for custom region.

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

nallath, 2020-09-04:
That is correct. The platform adhesion can only be set globally (and not per mesh / modifier mesh)

closed 2020-09-04 by nallath

Ghostkeeper, 2020-09-07:
Brim Distance is actually a per-object setting. Only it doesn't match the filter that Ncheltsov entered.