Ultimaker/Cura #8322: Text errors in the market

gailulun, 2020-09-04:
[image]
windows10
After the plug-in is installed in the marketplace, we need to exit Cura and restart it to make the plug-in take effect. When the plug-in is installed, this window will be displayed. The text on the button in the lower right corner of the window is wrong. The red box should have displayed "Ultimaker Cura". Now it shows "1%", which appears in version 4.6-4.7 in Chinese.When I changed the language back to English, the text displayed normally.
[image]

gailulun, 2020-09-05:
It seems that only simplified Chinese(简体中文) has this bug. If you change to traditional Chinese(正體字), it will be OK.
[image]

gailulun, 2020-09-05:
Similarly, the button that prompts you to restart Cura after you uninstall the plug-in is also wrong.
[image]

rburema, 2020-09-06:
I can see what happened here; `%1` is used as a placeholder in our strings (application name in this case). The translators are not supposed to change these, but it appears there's a separate unicode `%` character for this language that is different from the basic ASCII `%` character, that got inserted there instead.

I'll fix it for the next minor version (not the patch, sadly), and hope that this won't be repeated.

gailulun, 2020-09-07:
Thanks a lot :)

Ghostkeeper, 2020-09-07:
Thanks for the report! Slipped through, somehow.

commit f40b033c (2020-09-06): Don't use full-width unicode percent as placeholder in zh_CN locale.

Fixes #8322
commit 66132ef2 (2020-09-07): Don't use full-width unicode percent as placeholder in zh_CN locale.

Fixes #8322
closed 2020-09-07 by Ghostkeeper