Ultimaker/Cura #8323: [4.7] Start/End G-Code for printer is not properly updated with new version.

DrieStone, 2020-09-04:
**Application version**
4.7

**Platform**
Windows

**Printer**
Ender 5

**Reproduction steps**
1. Add Klipper code to start/end G-code (for instance: SET_PRESSURE_ADVANCE ADVANCE=0.736)
2. Update to 4.7
3. Look at printer settings.

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
Update process only copies part of the start/end g-code (seems to copy everything before an equals symbol)

**Expected results**
Copy the entire g-code to the new install

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
Here is my complete start g-code:

```
M201 X500.00 Y500.00 Z100.00 E5000.00 ;Setup machine max acceleration
M203 X500.00 Y500.00 Z10.00 E50.00 ;Setup machine max feedrate
M204 P500.00 R1000.00 T500.00 ;Setup Print/Retract/Travel acceleration
M205 X8.00 Y8.00 Z0.40 E5.00 ;Setup Jerk
M220 S100 ;Reset Feedrate
M221 S100 ;Reset Flowrate

SET_PRESSURE_ADVANCE ADVANCE=0.736
	
G28 ;Home
BED_MESH_CALIBRATE ;Bed Leveling
G92 E0 ;Reset Extruder
G1 Z2.0 F3000 ;Move Z Axis up
G1 X1 Y20 Z0.28 F5000.0 ;Move to start position
G1 X1 Y200.0 Z0.28 F1500.0 E15 ;Draw the first line
G1 X1.5 Y200.0 Z0.28 F5000.0 ;Move to side a little
G1 X1.5 Y20 Z0.28 F1500.0 E30 ;Draw the second line
G92 E0 ;Reset Extruder
G1 Z2.0 F3000 ;Move Z Axis up
machine_width = 225
```

4.7 changed it to:

```
M201 X500.00 Y500.00 Z100.00 E5000.00 ;Setup machine max acceleration
M203 X500.00 Y500.00 Z10.00 E50.00 ;Setup machine max feedrate
M204 P500.00 R1000.00 T500.00 ;Setup Print/Retract/Travel acceleration
M205 X8.00 Y8.00 Z0.40 E5.00 ;Setup Jerk
M220 S100 ;Reset Feedrate
M221 S100 ;Reset Flowrate

SET_PRESSURE_ADVANCE ADVANCE
```

Asterchades, 2020-09-05:
This is already known and already fixed for the upcoming (though not currently scheduled, to the best of my knowledge) 4.7.1 release.

Original report: https://github.com/Ultimaker/Cura/issues/8230
PR that fixes it: https://github.com/Ultimaker/Uranium/pull/633

bremco, 2020-09-06:
Taking a bit of extra time to get 4.7.1 properly 'notarized' so we don't annoy Apple users. Expect it in the coming few days.

closed 2020-09-07 by nallath