Ultimaker/Cura #8324: Left bar feature name when hovering

Liger0, 2020-09-04:
**Is your feature request related to a problem? Please describe.**
On the bar on the left where there are the objects management options there is no word about how they are called, so new people can have an hard time with some options like for example the support blocker.

**Describe the solution you'd like**
A popup saying the feature name.

**Affected users and/or printers**
Any

Ghostkeeper, 2020-09-07:
This is considered a bug. Because we used to have tooltips there, and they are still in the code too, but the tooltips stopped working. This is from 4.0 while hovering the mirror tool:
[image]

Liger0, 2020-09-07:
They work in 4.6.2 indeed, they got broken in 4.7

tinker3433, 2020-11-29:
Yes, This is very frustrating for me as a new user trying to first learn the software.

fieldOfView, 2020-11-29:
This was fixed in https://github.com/Ultimaker/Cura/pull/8731, which will be part of Cura 4.9

closed 2020-11-30 by nallath