Ultimaker/Cura #8325: Custom supports to work along with tree supports

Liger0, 2020-09-04:
**Is your feature request related to a problem? Please describe.**
Tree supports can be very useful, but some parts may still be no reached.
Using tree supports, custom supports aren't sliced.

**Describe the solution you'd like**
Allow the construction of custom supports along with tree supports

**Affected users and/or printers**
Any

Ghostkeeper, 2020-09-07:
Since it's expected behaviour that custom support meshes should still be printed even if tree support is used, I'll change this to a bug report.

Ghostkeeper, 2020-09-07:
However I'm unable to reproduce the issue. For me the custom supports still get printed as they should.
[image]
[image]

Ghostkeeper, 2020-09-07:
Keep in mind that if your custom support model has a great wide volume, you might want to increase the Support Infill Density, as it's usually set very low for tree support.

Liger0, 2020-09-07:
I looked up a bit and it seems this is in fact a feature request. I was wrong thinking they couldn't be printed together.
The reasons they weren't are different and caused by 2 facts:

First, I want tree support to start from the build plate, but I positioned the custom support to start from the model, so it isn't printed. 
The request here is to enable separate positioning options.

Second, when i positioned the custom support, it was on the model, so it wasn't created, but this, over not generating the custom support itself, caused another issue: in the area where it is supposed to be, the tree supports are skipped and the slicing takes an immense amount of time. 
So this would require a special check, so if it finds some intersection between the first layer of the custom support of the model and the tree supports with the supports from the build plate only, it should not considert he custom support but use the tree support instead of deleting it. What happens now is that it literally works as if the custom support was a support blocker instead.

Ghostkeeper, 2020-09-10:
> First, I want tree support to start from the build plate, but I positioned the custom support to start from the model, so it isn't printed.
The request here is to enable separate positioning options.

Indeed when "Touching Buildplate" is enabled, area support that doesn't touch the build plate is removed right now, including support meshes. It would be more powerful if that didn't happen. This is not specific to tree support; it also happens when you have "normal" support.

> in the area where it is supposed to be, the tree supports are skipped and the slicing takes an immense amount of time.

I'm unable to reproduce this issue. I have a support mesh set up like this:
[image]
And it ends up like this:
[image]
So the tree support keeps on generating through this custom support mesh. It also didn't take any longer to slice with the support blocker than without.

Liger0, 2020-09-10:
Maybe because the branches start from the same point? Mine was a single tree branch. Not sure.

Ghostkeeper, 2020-09-11:
Do you have a project file where support meshes cause tree support to be skipped, then?

closed 2024-12-07 by github-actions[bot]