Ultimaker/Cura #8326: Cura error when make in ubuntu: Note: xcb, opengl-desktop, system-freetype, fontconfig failed

JaideepBgit, 2020-09-05:
--------------------------------------------------------------------------------------------------------
--------------------------------------------------------------------------------------------------------
**Application version**
The application is taken from the below URL:
https://github.com/Ultimaker/Cura/wiki/Running-Cura-from-Source-on-Ubuntu
--------------------------------------------------------------------------------------------------------
--------------------------------------------------------------------------------------------------------
**Platform**
NAME="Ubuntu"
VERSION="20.04.1 LTS (Focal Fossa)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 20.04.1 LTS"
VERSION_ID="20.04"
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=focal
UBUNTU_CODENAME=focal
--------------------------------------------------------------------------------------------------------
--------------------------------------------------------------------------------------------------------
**Reproduction steps**
1. Clone the cura-build-environment repository.
2. Create a build subdirectory in that repository.
3. Open CMake and point its source directory to your cura-build-environment repository and the build directory to the build subdirectory.
4. Configure and generate. Use the GCC make files to compile these.
5. Open a terminal and run make in the build directory. If all goes well, this should build all dependencies. (FAILED HERE)
--------------------------------------------------------------------------------------------------------
--------------------------------------------------------------------------------------------------------
ERROR after following the below steps:
------------------------------------
Note: Also available for Linux: linux-clang linux-icc

ERROR: Feature 'xcb' was enabled, but the pre-condition 'libs.xcb' failed.

ERROR: Feature 'opengl-desktop' was enabled, but the pre-condition '(config.win32 && !config.winrt && !features.opengles2 && (config.msvc || libs.opengl))
                          || (!config.watchos && !config.win32 && libs.opengl)' failed.

ERROR: Feature 'system-freetype' was enabled, but the pre-condition 'features.freetype && libs.freetype' failed.

ERROR: Feature 'fontconfig' was enabled, but the pre-condition '!config.win32 && !config.darwin && features.system-freetype && libs.fontconfig' failed.

ERROR: The OpenGL functionality tests failed!
You might need to modify the include and library search paths by editing QMAKE_INCDIR_OPENGL[_ES2],
QMAKE_LIBDIR_OPENGL[_ES2] and QMAKE_LIBS_OPENGL[_ES2] in the mkspec for your platform.

Check config.log for details.
make[2]: *** [CMakeFiles/Qt.dir/build.make:110: Qt-prefix/src/Qt-stamp/Qt-configure] Error 3
make[1]: *** [CMakeFiles/Makefile2:826: CMakeFiles/Qt.dir/all] Error 2
make: *** [Makefile:84: all] Error 2
--------------------------------------------------------------------------------------------------------

**Expected results**
Should build all dependencies successfully

bremco, 2020-09-06:
Good grief, no need to shout!

While I realize this doesn't solve the problem, the document you linked to does offer an alternative to the 'first' method (and mentions that the automatic one is more error prone as well):

> Generally the CMake workflow is more automatic but more error-prone as well, while the alternative requires some more manual typing in the terminal.

> If this workflow doesn't work out for you, you can proceed below by installing the dependencies manually.

Instead, try from here: https://github.com/Ultimaker/Cura/wiki/Running-Cura-from-Source-on-Ubuntu#additional-dependencies

We're aware that we need to improve in this area, and I think we have already done some good work in the past. Currently, building Cura is ... not the most easy or robust process.

Ghostkeeper, 2020-09-07:
Maybe this should be reported to Qt rather than to us. All we're doing is to call the `configure` script of Qt with a bunch of options to disable things we don't need: https://github.com/Ultimaker/cura-build-environment/blob/master/projects/qt.cmake

Ghostkeeper, 2020-09-11:
I meant these: https://github.com/Ultimaker/Cura/wiki/Running-Cura-from-Source-on-Ubuntu#additional-dependencies

JaideepBgit, 2020-09-11:
Hi, thank you. I now understood.
link --> (https://github.com/Ultimaker/Cura/wiki/Running-Cura-from-Source-on-Ubuntu)--> link followed to perform the process:

1) Build environment *(step 1 in link mentioned above) is for building and installing the dependent libraries. It crashed at the above error [93%] in the title after lot of corrections. I was stuck and felt that this error should be solved, else it cannot be further proceeded. But it is not the case, I understood *(from your comment) the libraries for which the error is raised, can be installed separately. I have installed the libraries separately.
2) Then I have proceeded to the build of the CURA *(step 7 in link mentioned above) to produce executable file. It crashed with another error saying it couldn't create executable in cx_freeze due to the unavailability of libprotobuf.so.3.9.2.0. I skipped this and proceeded now with installing other libraries separately.
3) And later the CURA itself and Ultimaker CURA is working now.
4) Blackbelt CURA has a problem, for which I'll open a fresh issue.
Thank you @Ghostkeeper

closed 2020-09-11 by JaideepBgit