Ultimaker/Cura #8327: OBJ file with "negative vertex reference numbers" does not load

ScottFerg56, 2020-09-05:
**Application version**
Cura 4.7.0

**Platform**
Windows Version 10.0.18363 Build 18363

**Printer**
not relevant

**Reproduction steps**
1. Loaded OBJ file "Box FM.obj" (attached as a TXT file)

**Screenshot(s)**
Just a Warning message: "The selected model was too small to load."

**Actual results**
The above warning message.

**Expected results**
The OBJ model should load, a simple 20x20x5 box.

**Log file**
n/a

**Additional information**
This is perhaps an obscure feature of the OBJ file format; but for completeness would likely be easy enough to support.
Faces (lines beginning with 'f') may use "negative vertex reference numbers". (See http://www.martinreddy.net/gfx/3d/OBJ.spec). In this file, for example, a triangular face is represented by the sequence:
v 0 0 5
v 0 20 5
v 20 0 5
f -1 -2 -3
where the face references the previous 3 vertices with negative indexes.
I suspect this feature is not supported, resulting in the failure to load the file.
FWIW, Fusion 360 and the Windows 3D Viewer both load the file successfully.

[Box FM.txt](https://github.com/Ultimaker/Cura/files/5178532/Box.FM.txt)

JaideepBgit, 2020-09-06:
ScottFerg56. How did you install CURA? Did you build from source or installed using .exe file? I'm trying to do build from source and it is not building

ScottFerg56, 2020-09-06:
I was using the official download website link, as provided by Cura during notification of a new version. NOT a local build.