Ultimaker/Cura #8329: When using "One at a Time", "Arrange all models" doesn't maximise space usage

Mearman, 2020-09-06:
Arranging when using "one at a time" print sequence leaves unnecessary space between models and does not place models near enough to the edge of the build plate

Ghostkeeper, 2020-09-07:
Indeed, the collision areas are wrong in one-at-a-time mode. The effect of that is also visible when enabling the "Ensure models are kept apart" preference.
We had already discovered this internally and have a bug about it in our backlog (devs: see CURA-6833).

daton89, 2020-10-12:
In my case it puts only one model in the center and all the others outside of the bed. 
But it works ok when I use `All at once` as _Print Sequence_

Taomyn, 2021-01-30:
Any ETA on when this will be fixed?

Ghostkeeper, 2021-02-05:
Hard to say. We haven't planned it in for a specific release. It's about halfway down our priority list.

closed 2024-12-07 by github-actions[bot]