Ultimaker/Cura #8332: [Request] Are G2/G3 Commands in Cura's Future?

KenLucke, 2020-09-06:
I am sure it has been brought up before, but is there any future plans for implementing optional G2/G3 (arc) commands in Cura's gcode output?  It would help eliminate some of the blobbing/zitting/stuttering that some 8-bit machines suffer when getting their serial buffer overrun by an infinitesimal number of tiny short segments to create a circle/arc, reduce gccode size, and make printing more efficient.

ClockeNessMnstr, 2020-09-06:
you may be suffering from #8321 ? this issue just showed me the door....

But if you're interested, I made a post processing script that rebuilds arcs into larger segments of G2/3 commands. 
However, as is, it does not validate extrusion volume per path length (mm^3)/mm if it were to change along an arc.
I believe marlin essentially converts these to G1 segments again though. if that's the case it still may be fairly faceted but should stop stuttering. 

[RebuildCircles.zip](https://github.com/Ultimaker/Cura/files/5180191/RebuildCircles.zip)

KenLucke, 2020-09-06:
There's an OctoPrint plugin (ArcWelder) that does this, fairly well, but it's all still post-processing and cobbled guesswork.  If the base code were generated WITH the G2/G3 commands calculated in the process, all that would be unnecessary.  A simple checkbox in the Printer Settings to enable/.disable G2/G3 processing for the printer in Cura would solve the excuse I've seen before that "not all printers support G2/G3" (in reality, I think it's the firmware, not the printer, but that's a nitpick)

ClockeNessMnstr, 2020-09-06:
As you didn't mention a file format, I'm assuming you're working with .STL files. no matter how high the resolution is on your STL there are actually no curves. To my knowledge there's no options to have curved data in .STL
It could be made a setting in cura instead of a post process but it will be the same math, not actually any different conversion.

KenLucke, 2020-09-06:
If there was a reason to switch to something other than .stl, for instance if the slicer was compatible, I'd do it in a heartbeat :)

Asterchades, 2020-09-07:
3MF (3D Manufacturing Format) was supposed to replace STL for 3D printing and, at least theoretically, supports arcs; among other features (multi-material/-colour, preview images, notes, etc). But it hasn't really taken off due to how embedded STL already is within the industry as a result of how long it's existed. It's a bit like Arduino processors in printers: people use them because that's what has been established as working, rather than what's the best option.

Like Arduinos giving way to Arms, though, 3MF may well eventually take over from STL. But it may require 3D printing to become a little more mainstream first to drive the desire for such extended features, since "it works" is a pretty good reason not to rock the boat.

fieldOfView, 2020-09-07:
> 3MF (...) at least theoretically, supports arcs

No, not really. Meshes in 3MF files are polygonal, and polygons don't have arcs. Sure, 3MF files support arbitrary metadata (such as the settings that are stored in a Cura project), and you could encapsulate something that supports curves in that metadata. But you would not be using the 3MF specification and Cura would not be able to read it nor would there be any 3d package be able to edit/write it. 

Neither Cura nor any other fdm-slicer I know supports slicing non-polygonal objects. Slicers need to be able to make intersections of models, and they are all architecturally limited to making and manipulating intersections of polygonal models.

Ghostkeeper, 2020-09-08:
This is a duplicate of https://github.com/Ultimaker/CuraEngine/issues/429.

There are some file formats that support curved surfaces. 3MF is not one of them, although it is an extensible file format and an extension for nurbs could be defined. OBJ does support various kinds of curved surfaces (but most applications incl. Cura currently ignore them). In reality, curved objects are currently in the domain of CAD software only.

From experiments we've seen that Marlin's support for G2/G3 is pretty basic. This is not a priority for us because of the difficulty of implementation and the estimated minimal benefits from it.

closed 2020-09-08 by Ghostkeeper

uis246, 2022-02-01:
CSG and STEP are supporting arcs. CSG actually supports spheres and cylinders, STEP at least cylinders and most likely spheres too.