Ultimaker/Cura #8334: Generic PC material does not have supported profiles on Creality

amantegazzi, 2020-09-06:
**Application version**
4.7

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)
Win10 Pro 64bit, Core i7-6700, Radeon R9

**Printer**
Creality CR10S Pro

**Reproduction steps**
1. Select Generic PC as material, result in not supported profile

**Screenshot(s)**
[image]

**Actual results**
Cannot have multiple profiles for standard and low quality.

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
NA

**Log file**
NA

**Additional information**
(Extra information relevant to the issue.)

nallath, 2020-09-07:
We rely on community contributions for third party printers. Since no-one contributed these profiles, Cura shows a not supported profile warning.

Do note that you can still manually set the layer height to be something lower. The not supported warning is just there to indicate that Cura fell back to more generic defaults, which may or may not work (aka; A significant amount of tweaking might be needed to get it to work).

If any such profiles for PC are contributed, we would be more than happy to review / merge them.

closed 2020-09-07 by nallath

amantegazzi, 2020-09-16:
Is there a documented way to contribute?

Best, Andrea

Closed #8334<https://github.com/Ultimaker/Cura/issues/8334>.